couriersud/msigd #1: Prestige PS341WU

Wehin, 2020-02-11:
./msigd --info --debug --query
DEBUG: Initializing HID lib
Unexpected unknown01 : 5b00
Unexpected unknown04 : 5b00
Unexpected unknown05 : 5b00
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:  0x1462
Product Id: 0x3fa4
Product:    MSI Gaming Controller
Serial:     A02019010700
DEBUG: Special 01 b0 : 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 35 38 30 30 31 39 30 0d : 01 35 62 30 30 00 00 00 30 30 30 0d 00 00 00 00 
msigd: Error querying device on power - got <5b00>
Try 'msigd --help' for more information.
unknown02 : 001
msigd: Error querying device on mode - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on unknown03 - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on frequency - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on game_mode - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown06 - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on response_time - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on enable_dynamic - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on hdcr - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_display - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_position - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on alarm_clock - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_index - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_time - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on alarm_clock_position - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on screen_assistance - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown08 - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on screen_size - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on night_vision - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on pro_mode - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on eye_saver - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on image_enhancement - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on brightness - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on contrast - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on sharpness - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on color_preset - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on red - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on green - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on blue - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on rgb - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown09 - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on input - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on pip - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pip_input - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on pbp_input - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on pip_size - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on pip_position - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_language - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_transparency - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_timeout - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on sound_enable - got <5b00>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on back_rgb - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_up - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_down - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_left - got <5b00>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_right - got <5b00>
Try 'msigd --help' for more information.

couriersud, 2020-02-11:
This looks weird. The return values are truncated. 
Can you please try 
```
make clean
make USE_HIDAPI=0
```
and try 
```
./msigd --info --debug --query
```
again? 
If this doesn't help I need to improve error logging.

Wehin, 2020-02-11:
# ./msigd --info --debug --query
```
DEBUG: Initializing usb_lib
DEBUG: Scanning USB devices...
DEBUG: Found device MSI Gaming Controller 

DEBUG: Interface 0 alt 0 class 3
DEBUG: Endpoint 01 0040
DEBUG: Endpoint 1 does is not of type output
DEBUG: Endpoint 02 0040
DEBUG: Endpoint 2 does is not of type input
DEBUG: Found 2 interfaces, using interface 0
DEBUG: Setting Configuration
DEBUG: failed to detach kernel driver from USB device: -22
DEBUG: Releasing interface 0

DEBUG: Error 1 releasing Interface 

msigd: No usb device found
Try 'msigd --help' for more information.
```

couriersud, 2020-02-11:
Short solution since this is for debugging only:

Please try
```
sudo ./msigd --info --debug --query
```

The long story is in section 4.1. of the README.MD.

I will commit an update in a couple of minutes which has more flexible return code reading. This will ignore zero bytes.  The monitor should be supported by the code, at least that is my conclusion from looking at the CreatorOS exe and the first packets it tries to exchange with my monitor.

couriersud, 2020-02-11:
Committed code update as promised.

Wehin, 2020-02-11:
```
# ./msigd --info --debug --query
DEBUG: Initializing HID lib
Unexpected unknown03 : 5b0000?
Unexpected unknown01 : 5b00V06
Unexpected unknown04 : 5b00DA1A019340045
Unexpected unknown05 : 5b00000
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:  0x1462
Product Id: 0x3fa4
Product:    MSI Gaming Controller
Serial:     A02019010700
DEBUG: Special 01 b0 : 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 35 38 30 30 31 39 30 0d : 01 35 62 30 30 00 00 00 30 30 30 0d 00 00 00 00 
msigd: Error querying device on power - got <5b00001>
Try 'msigd --help' for more information.
unknown02 : 000
msigd: Error querying device on mode - got <5b00013>
Try 'msigd --help' for more information.
msigd: Error querying device on frequency - got <5b00030>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on game_mode - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown06 - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on response_time - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on enable_dynamic - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on hdcr - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_display - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_position - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on alarm_clock - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_index - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_time - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on alarm_clock_position - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on screen_assistance - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown08 - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on screen_size - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on night_vision - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on pro_mode - got <5b00003>
Try 'msigd --help' for more information.
msigd: Error querying device on eye_saver - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on image_enhancement - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on brightness - got <5b00075>
Try 'msigd --help' for more information.
msigd: Error querying device on contrast - got <5b00050>
Try 'msigd --help' for more information.
msigd: Error querying device on sharpness - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on color_preset - got <5b00002>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on red - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on green - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on blue - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on rgb - got <5b00���>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown09 - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on input - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on pip - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pip_input - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on pbp_input - got <5b00002>
Try 'msigd --help' for more information.
msigd: Error querying device on pip_size - got <5b00002>
Try 'msigd --help' for more information.
msigd: Error querying device on pip_position - got <5b00001>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_language - got <5b00001>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_transparency - got <5b00000>
Try 'msigd --help' for more information.
msigd: Error querying device on osd_timeout - got <5b00020>
Try 'msigd --help' for more information.
msigd: Error querying device on sound_enable - got <5b00000>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on back_rgb - got <>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_up - got <5b00002>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_down - got <5b00003>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_left - got <5b00004>
Try 'msigd --help' for more information.
msigd: Error querying device on navi_right - got <5b00005>
Try 'msigd --help' for more information.
```

couriersud, 2020-02-11:
Commited an update which ignores zero bytes at pos 5,6,7. This should fix most of the above error messages.
Note: you should be able to set monitor osd parameters now.

Wehin, 2020-02-11:
tip: for fedora 
```
dnf install libusb-devel hidapi-devel
```
yep, some parameters :+1: 
```
# ./msigd --info --debug --query
DEBUG: Initializing HID lib
Unexpected unknown03 : 00?
Unexpected unknown01 : V06
Unexpected unknown04 : DA1A019340045
Unexpected unknown05 : 000
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:  0x1462
Product Id: 0x3fa4
Product:    MSI Gaming Controller
Serial:     A02019010700
DEBUG: Special 01 b0 : 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 35 38 30 30 31 39 30 0d : 01 35 62 30 30 00 00 00 30 30 30 0d 00 00 00 00 
power : on
unknown02 : 001
mode : designer
frequency : 030
DEBUG: Error 0 reading from HID device
msigd: Error querying device on game_mode - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown06 - got <>
Try 'msigd --help' for more information.
response_time : normal
DEBUG: Error 0 reading from HID device
msigd: Error querying device on enable_dynamic - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on hdcr - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_display - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on refresh_rate_position - got <>
Try 'msigd --help' for more information.
alarm_clock : off
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_index - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_time - got <>
Try 'msigd --help' for more information.
alarm_clock_position : left_bottom
screen_assistance : 0
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown08 - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got <>
Try 'msigd --help' for more information.
screen_size : 19
DEBUG: Error 0 reading from HID device
msigd: Error querying device on night_vision - got <>
Try 'msigd --help' for more information.
pro_mode : designer
eye_saver : off
image_enhancement : off
brightness : 75
contrast : 50
sharpness : 0
color_preset : warm
DEBUG: Error 0 reading from HID device
msigd: Error querying device on red - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on green - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on blue - got <>
Try 'msigd --help' for more information.
rgb : -1575756
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown09 - got <>
Try 'msigd --help' for more information.
input : hdmi1
pip : off
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pip_input - got <>
Try 'msigd --help' for more information.
pbp_input : dp
pip_size : large
pip_position : right_top
osd_language : 1
osd_transparency : 0
osd_timeout : 20
sound_enable : off
DEBUG: Error 0 reading from HID device
msigd: Error querying device on back_rgb - got <>
Try 'msigd --help' for more information.
navi_up : game_mode
navi_down : screen_assistance
navi_left : alarm_clock
navi_right : input
```

Wehin, 2020-02-11:
```
# ./msigd --pro_mode user
Unexpected unknown03 : 00?
Unexpected unknown01 : V06
Unexpected unknown04 : DA1A019340045
Unexpected unknown05 : 000
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
```

Supported monitor modes:
-  User (equal "user")
-  Adobe RGB  (return by query "reader")
-  Display P3 ("cinema")
-  sRGB  ("designer")
-  HDR ( query segfault)
-  Cinema  (blank)
-  Reader (blank)
-  Black-White (blank)
-  DICOM (blank)
-  EyeCare (blank)

couriersud, 2020-02-11:
Cool. We are getting there.  I'll include the fedora tip in the readme.md.
The rest needs more consideration. I will not be able to do this today but expect an update till the weekend.

`DEBUG: Error 0 reading from HID device` 
indicates that less than PACKET_SIZE (64) bytes were returned by hid driver. This must not be an error. Unfortunately it triggers non-trivial code changes.
You should be able now to set osd parameters which are supported in the output above.

Thanks a lot for your support. Much appreciated. Your feedback will allow me to now introduce multi-monitor support.
I'll leave a comment here once I made the changes.

couriersud, 2020-02-11:
Is this the order of modes in monitor OSD? 
I need to extend the functionality now to support different parameter options depending on detected monitor.

Wehin, 2020-02-11:
yep, its from OSD

couriersud, 2020-02-11:
Ok, 10 modes, in the order given. 
As said, I need to extend the code to support per-monitor value lists. 
Not a big deal, but needs some time.
In the mean time, any more information/lists from the OSD are appreciated.

Wehin, 2020-02-11:
already working on it :)

it would be nice to add language support in the output, I can translate into Russian

Wehin, 2020-02-11:
**Professional**
```
 - Mode *
    - User
    - Adobe RGB
    - Display P3
    - sRGB
    - HDR
    - Cinema
    - Reader
    - Black-White
    - DICOM
    - EyeCare
 - Alarm Clock *
     - Location 
        - Left Top
        - Right Top
        - Left Bottom
        - Right Bottom
     - OFF
     - 15:00 **
     - 30:00 **
     - 45:00 **
     - 60:00 **
 - Screen Assistance
    - OFF
    - Ruler:Center
    - Ruler:Edge
    - Scale:Vertical
    - Scale:Horizontal
    - Reference Line:Vertical
    - Reference Line:Horizontal
    - Grid
    - Rule of Thirds
    - 3D motion sickness assistance 

Tip:
 * Alarm have 4 prests.
 ** default value for preset, value can be changed. max value is 99:59.
```

**Image**
```
 - Brightness [0..100]
 - Contrast [0..100]
 - Sharpness [0..5]
 - Gray Level [0..20]
 - Image Enhancement
   - OFF
   - Weak
   - Medium
   - Strong
   - Strongest
 - Responce Time
    - Normal
    - Fast
    - Fastest
 - Low Blue Light [on | off]
 - Local Dimming [on | off]
 - Color Temperature
    - Standart
        - 5000K
        - 5500K
        - 6500K
        - 7500K
        - 9300K
        - 10000K
    - Customization
        - Red [0..100]
        - Green [0..100]
        - Blue [0..100]
 - Gamma [1.8 | 2 | 2.2 | 2.4 | 2.6]
 - Saturation
   - Red [0..100]
   - Green [0..100]
   - Blue [0..100]
   - Cyan [0..100]
   - Magenta [0..100]
   - Yellow [0..100]
 - Hue
   - Red [0..100]
   - Green [0..100]
   - Blue [0..100]
   - Cyan [0..100]
   - Magenta [0..100]
   - Yellow [0..100]
 - Zoom In [on | off]
   - Location
     - Center
     - Up-Left
     - Up-Right
     - Down-Left
     - Down-Right
 - Screen Size
    - Auto
    - 4:3
    - 16:9
    - 21:9
    - 1:1

Tip:
* Switching the mode makes some parameters inaccessible to change.
```    
**Input Source**
```
- HDMI1
- HDMI2
- DP
- Type C
```

Wehin, 2020-02-12:
Done

couriersud, 2020-02-12:
Committed a new version for you to test. This one will identify the monitor series (MAG/PS) and use different osd parameters and value lists. Not all of your feedback has been used yet.
The purpose of this version is to identify query codes (e.g. 004=0) which are used by the PS series.

Please provide the `./msigd --info --debug --query` output

Wehin, 2020-02-12:
```
./msigd --info --debug --query
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: PS Series
DEBUG: Special 01 b0 : 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 35 38 30 30 31 39 30 0d : 01 35 62 30 30 00 00 00 30 30 30 0d 00 00 00 00 
power : on
unknown02 : 000
serial : DA1A019340045
frequency : 030
unknown05 : 000
response_time : normal
alarm_clock : off
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_index - got <>
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_time - got <>
Try 'msigd --help' for more information.
alarm_clock_position : left_bottom
screen_assistance : 0
screen_size : auto
pro_mode : dcpi_p3
eye_saver : off
image_enhancement : off
brightness : 75
contrast : 50
sharpness : 0
color_preset : 6500K
rgb : 100100100
unknown460 : 010
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown470 - got <>
Try 'msigd --help' for more information.
unknown480 : 000
unknown490 : 000
unknown4<0 : bbb
unknown4=0 : 000
unknown4;0 : bbb
unknown4:0 : 002
unknown4<1 : bbb
unknown4=1 : 000
unknown4;1 : bbb
input : hdmi1
pip : off
pbp_input : hdmi1
pip_size : large
pip_position : right_top
unknown670 : 000
unknown680 : 000
unknown690 : 000
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown69: - got <>
Try 'msigd --help' for more information.
osd_language : 1
osd_transparency : 0
osd_timeout : 30
quick_charge : off
audio_source : analog
navi_up : pro_mode
navi_down : screen_assistance
navi_left : alarm_clock
navi_right : input
```

couriersud, 2020-02-12:
Thanks a lot. Most settings are supported now.

- Setting alarm_index and clock_time seems not to be supported. Perhaps it is a write only feature, try setting an alarm clock using  
`/msigd --alarm_clock_index 3 --alarm_clock_time 55 --alarm_clock 3`
- for the "unknownXXX" you need to find out what osd setting is related to them. A value of "bbb" indicates RGB or color values, the 670-690 settings should be PIP/PBP related.
- "470" and "69:" are not supported (timeout)

Wehin, 2020-02-12:
**Counter**

Work:
``` ./msigd --alarm_clock [1..4, off] ```

Not work
```
./msigd --alarm_clock_index 1 --alarm_clock_time 4444
msigd: Error setting --alarm_clock_index
```

```
./msigd --alarm_clock_index 3 --alarm_clock_time 55 --debug
DEBUG: Initializing HID lib
DEBUG: Setting alarm_clock_index to 003
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --alarm_clock_index
Try 'msigd --help' for more information.
```

Wehin, 2020-02-12:
**power**

power on  -> display is off
```
./msigd --power on --debug 
DEBUG: Initializing HID lib
DEBUG: Setting power to 001
DEBUG: Got unexpected return <_______________________________________________________________>
msigd: Error setting --power
```

power off  -> display is off
```
 ./msigd --power off --debug
DEBUG: Initializing HID lib
DEBUG: Setting power to 000
DEBUG: Got unexpected return <_______________________________________________________________>
msigd: Error setting --power
Try 'msigd --help' for more information.
```

"power on" when display is off
```
./msigd --power on --debug
DEBUG: Initializing HID lib
msigd: No usb device found
Try 'msigd --help' for more information.
```

try to wait auto suspend and resume on command

couriersud, 2020-02-12:
Only "--power off" is working. Same here. This is expected. When the monitor is off, the usb interface is off as well. I was disappointed as well.
However on MAG321Curv "--power on" doesn't turn off the monitor (what I would expect). 
And the unexpected return is not what I would expect. 

Alarm setting: Is it possible to set the alarms in the Windows application (CreatorOSD.exe)?

Wehin, 2020-02-12:
no have Windows sorry.

more results
```
--power                 ?broken
--response_time         +done
--alarm_clock           +done
--alarm_clock_index     -fail
--alarm_clock_time      -fail
--screen_assistance     *confuse only 9 presets + off, set 10 == 0 (off), 11 == 1, 12 == 2
--eye_saver             *confuse do nothing
--image_enhancement     +done
--brightness            +done
--contrast              +done
--sharpness             +done
--rgb                   *confuse current values is 100100100 > max
--unknown10             =miss
--input                 +done tip: main source in pip mode
--pbp_input             *confuse `pip_source`
--pip_size              +done
--pip_position          +done
--toggle_display        *confuse no effect `off` value, `on` toggle primary
--toggle_sound          *confuse currently toggle to `digital sound` an select audio source 0|1 in pip/pbp. ?Rename to audio_source HDMI1=0, HDMI2=1, DP=2, USBC=3
--osd_language          *confuse have 29 language(0..28) values limit 20
--osd_transparency      +done
--osd_timeout           +!done tip: 1..30 seconds, 0 == infinity :)
--reset                 *confuse no effect `off` value
```

Wehin, 2020-02-12:
request:
result some value for scripting
```
./msigd --osd_language
-> 2
```
additional flag `-v` result values with description for human

couriersud, 2020-02-12:
power: as indicated, only power off working, on PS series looks like touching it means power off
alarm-clock index and time: I have a workaround, this will take time to implement.
screen assistance: on my todo list
eye saver: will limit to MAG series.
rgb: 100100100 means red=100, green=100, blue=100
        050050050 means red=50,green=50,blue=50
unknown10: what does "=miss" means?
-pbp_input: need to think about this.
- toggle_sound: I don't understand your comments. This is a write only setting.
Both toggle_sound and toggle_display only support "on". Need to document this. 
- reset - another setting which only works with "on".
I will add these comments to the "--help" output.

couriersud, 2020-02-12:
> request:
> result some value for scripting
> 
> ```
> ./msigd --osd_language
> -> 2
> ```
> 
> additional flag `-v` result values with description for human

Please open another issue for this.

Wehin, 2020-02-12:
rgb: yep, limit 100100100 > limit 100000000
```
./msigd --rgb 100100100
msigd: Unknown value <100100100> for option rgb
```

toggle_sound: toggle `Settings:Audio Source` to digital (if was selected `Analog`)and currently `off` change to source 0(HDMI1) and `on` change to source 1(HDMI2), we have 2 more sources  2 and 3.

unknown10: works but it’s not known what does

reset and power: better to use without additional arguments

Wehin, 2020-02-12:
maybe rename `rgb` to `color_rgb`? like `color_preset`

couriersud, 2020-02-13:
> maybe rename `rgb` to `color_rgb`? like `color_preset` or combine them `color_temperature` with values [5000K,5500K,6500K,7500K,9300K,10000K, (0..100,0..100,0..100)]
> 
> Another issue ? :)

No, this would introduce endless if statements in the code and make adding further monitors a lot more difficult.

couriersud, 2020-02-13:
> rgb: yep, limit 100100100 > limit 100000000
> 
> ```
> ./msigd --rgb 100100100
> msigd: Unknown value <100100100> for option rgb
> ```
Fixed.
> 
> toggle_sound: toggle `Settings:Audio Source` to digital (if was selected `Analog`)and currently `off` change to source 0(HDMI1) and `on` change to source 1(HDMI2), we have 2 more sources 2 and 3.

Changed toggle_sound to accept hdm1,hdmi1,dp,usbc
> 
> unknown10: works but it’s not known what does
> 
I'll leave this in. Some time we may understand it
> reset and power: better to use without additional arguments

That would complicate the logic and scalability a lot. Will reconsider later.

couriersud, 2020-02-13:
Closing this issue now. Please open a new issue if new bugs surface or new functionality is discovered.

closed 2020-02-13 by couriersud