couriersud/msigd #2: Additional functionality

Wehin, 2020-02-12:
additional flags:
-v result values with description for human
-s silent and result some value for scripting

example:
```
./msigd -s --osd_language
-> 2
``
or
```
 ./msigd -v --contrast
-> Contrast: 50  Min:0 Max:100
```

Wehin, 2020-02-14:
flags for key_macro e.g.
`-m` check only key_macro without other stuff, return 1 when pressed
`-w 100`  loop wait press key_macro, check every 100ms, exit loop with result 1 when pressed.

couriersud, 2020-04-19:
```
./msigd  -n -f pip
pip : 0
```
and
```
./msigd  --wait macro_key=pressed
```
should address the  issues raised here.
@Wehin Ok to close?

Wehin, 2020-04-23:
how oneshot set and return current set value?
```
/msigd --brightness 1001; echo $?
msigd: Unknown value <1001> for option brightness
Try 'msigd --help' for more information.
1
```
ok, code is 1 with message

```
./msigd --brightness 100; echo $?
0
```
code is 0 without message

```
./msigd --brightness 100 -n -f brightness; echo $?
brightness : 10
0
```
code is 0 with older settings, broken order ?

couriersud, 2020-04-24:
Closing. Order was fixed.

closed 2020-04-24 by couriersud