couriersud/msigd #3: PS Series options

Wehin, 2020-02-13:
```
--alarm_clock_position  +!done       hmmm... `custom` is set... how use `custom`?
--screen_assistance     *confuse     values in order: `off`, `ruler_center`, `ruler_edge`, `paper_sizes_vertical`, `paper_sizes_horizontal`, `center_line_vertical`, `center_line_horizontal`, `grid`, `thirds`, `3d_assistance`
--screen_size           *confuse     it's not size, `scale_aspect_ratio` or `screen_scale`
--pro_mode              +!done       fix type `dcpi_p3` => `dci_p3`; do not have `cal*` modes, `cal1` switch to `user`, `cal2` => `adobe_rgb`, `cal3` =>`dci_p3`
--color_preset          +done        maybe rename `custom` to `color_rgb`
--pip                   +done
--pip_input             +done
--toggle_sound          +done
--osd_language          +done
--quick_charge          ?broken      it's not quick charge, it's “Information On Screen”, `screen_info`?
--audio_source          +done
--navi_up               +done
--navi_down             +done
--navi_left             +done
--navi_right            +done
```

Wehin, 2020-02-13:
typo in result
power : **-** on

```
 ./msigd  --query
power : -on
unknown110 : 001
serial : DA1A019340045
frequency : 030
unknown190 : 000
msigd: Error querying device on unknown1@0 - got <>
Try 'msigd --help' for more information.
```

couriersud, 2020-02-13:
> typo in result
> power : **-** on

Will remove power from query results. No power ==> tool fails

couriersud, 2020-02-13:
> --screen_size           *confuse     it's not size, `scale_aspect_ratio` or `screen_scale`
Manual [here](https://download.msi.com/archive/mnu_exe/monitor/PS341v1.0_English.pdf) calls it "Screen Size". They do this in all manuals. I wanted to be consistent with the manuals.

> --screen_assistance     *confuse     values in order: `off`, `ruler_center`, `ruler_edge`, `paper_sizes_vertical`, `paper_sizes_horizontal`, `center_line_vertical`, `center_line_horizontal`, `grid`, `thirds`, `3d_assistance`

Manual (see link) seems to be wrong. Fixed.

couriersud, 2020-02-13:
> --alarm_clock_position  +!done       hmmm... `custom` is set... how use `custom`?

I have no idea. How would you change it from the monitor OSD? There are some things for which the USB communication needs to be logged between CreatorOSD app and the monitor. (Virtual Machine and Wireshark are your friends).

Wehin, 2020-02-13:
yep... in some places the documentation is not consistent

Wehin, 2020-02-13:
They don't understand what they're talking about.
"screen size" is not the same as "scale aspect ratio"

couriersud, 2020-02-13:
On the MAG321Curv you can select 19 or 24 here which will simulate a 19 inch or 24 inch monitor. The *visible* size changes. Visible size also is limited with 4:3 option. On the PS34WU you are right, only aspect ratios. On the MAG series, also absolute size (19,24)

Wehin, 2020-02-13:
19 or 24 it's scale ;)

Wehin, 2020-02-14:
As a fact:
a1,a2,a3,a4 -- set the appropriate timer
n - starts and displays a timer
if `n` == 0, then the timer continues to display but pauses.

couriersud, 2020-02-14:
On MAG321CURV n==0 removes timer from display.

Wehin, 2020-02-14:
```./msigd  --alarm4x 10,20,30,40,1```

and next

```./msigd  --alarm4x 0,20,30,40,0``` 

preset_1 set to 00:00, continues displayed 00:06 :)

Wehin, 2020-02-14:
this is not a problem, hiding the "--alarm_clock off" timer works great

Wehin, 2020-02-14:
tip: to README table PS341WU current displayed firmware is  FW.024

couriersud, 2020-02-14:
Closing this since all topics resolved.

closed 2020-02-14 by couriersud