couriersud/msigd #4: Additional Key

Wehin, 2020-02-13:
Step 1
`./msigd  --query --info --debug`
result `unknown110 : 000`

Step 2
Press key

Step 3
Wait any time

Step 4
./msigd  --query --info --debug
result `unknown110 : 001`

Step 5
./msigd  --query --info --debug
result `unknown110 : 000`

Resume:
I think “unknown110” means “001” if the button was pressed before checking and reset button state. For sure there is also the ability to configure the action at press of a button

couriersud, 2020-02-13:
Changed this settting to "macro_key" and made it read-only.
To check for it you need to constantly query the monitor, i.e.
`while true; do msigd --query | grep macro_key; done`
That is exactly how the Windows GamingOSD app checks for the key. MAG321Curv does not have such a key. Good finding.

Wehin, 2020-02-13:
msigd --query does a lot of stuff, need a simpler verification algorithm. It can be done through a key or a separate utility.

couriersud, 2020-02-13:
Options for msigd will get an overhaul at some time.

Wehin, 2020-02-14:
unknown490 => `local_dimming`

Wehin, 2020-02-14:
unknown4=0 => `zoom`

unknown4=1 => `zoom_loaction`
values in order: `center`, `left_top`,`right_top`,`left_bottom`,`right_bottom`

Wehin, 2020-02-14:
unknown460 => `gray_level` 
values 0..20

Wehin, 2020-02-14:
unknown480 => `low_blue_light` toggle 0,1

Wehin, 2020-02-14:
unknown4:0 => `gamma`
values: [1.8 | 2 | 2.2 | 2.4 | 2.6]

Wehin, 2020-02-14:
unknown4;0 => `saturation_rgb`
unknown4;1 => `saturation_cmy`

`unknown4;0 : bb~` => 50,50,78
50,50,80 => segfault

Wehin, 2020-02-14:
unknown4<0 => `hue_rgb`
unknown4<1 => `hue_cmy`

values like `saturation_*`

Wehin, 2020-02-14:
unknown670 => `pbp1_source`
unknown680 => `pbp2_source`
unknown690 => `pbp3_source`
`pbp4_source` can't find, not all unknown is reported?

Wehin, 2020-02-14:
`hue_cmy : 50050050`, maybe need formatted output with lead , like => `hue_cmy : 050050050`

couriersud, 2020-02-14:
> unknown670 => `pbp1_source`
> unknown680 => `pbp2_source`
> unknown690 => `pbp3_source`
> `pbp4_source` can't find, not all unknown is reported?

perhaps 00620 - pip_input?

Wehin, 2020-02-14:
no pip it's pip

Wehin, 2020-02-14:
Tip: power button is software controlled,
set `alarm_clock_position : custom` and navigate to OSD alarm:location => freez OSD,  no reaction power button :)

Wehin, 2020-02-14:
way to set `unknown190` i think i'ts macro_key function

Wehin, 2020-02-14:
no keys for `pbp4_source` and `quick_charger`

Wehin, 2020-02-14:
`00120` is equal `00300`
`unknown190` => return `000`
`unknown310` => return `000`

`quick_charge` can't find.

couriersud, 2020-02-14:
Thanks a lot, should all be fixed now.

Wehin, 2020-02-14:
```msigd: Error setting --unknown190```

hmm..unknown readonly

Wehin, 2020-02-14:
I looked for something about `quick_charge` and found in the file `MonitorControlLibrary.dll` entries similar to commands.

note: command order is  0 1 2 3 4 5 6 7 8 9 : ; < =
because  ":"  = 0x3A ";" == 0x3B  "<" == 0x3C  "="==0x3D

with lead 5800
```
120
130
140
150
160
170
190
1@0

200
210
220
230
240
250
251
260
261
262
263
270
271
280
290
2:0
2;0

300
310
340

400
410
420
430
431
432
433
434
435
460
470
480
490
4:0
4;0
4;1
4<0
4<1
4=0
4=1

500

600
610
620
630
640
670
680
690
6:0

800
810
820
840
850

900
910
920
930
```

with lead 5b00
```
1f
1g
1h
2>
2<
6<

100
120

200
210
220
230
240
250
251
260
261
262
263
270
271
280
290
2:0
2;0
2=0

300
310
340

400
410
420
430
431
432
433
434
435
440
460
470
480
490
4:0
4;0
4;1
4<0
4<1
4=0
4=1

500

600
610
620
630
640
650
660
6;0

800
810
820
840
850
860

900
910
920
930
```

couriersud, 2020-02-14:
All of those should have been covered by us with the exception of
```
1f
1g
1h
2>
2<
6<
```
`1f` is alarm4x. The others are "multi" setting commands as well and used by the GamingOSD/CreatorOSD apps. 
`1g` is e.g. used when you change pip/pbp from the app.

Wehin, 2020-02-14:
yep, perhaps there is something interesting among them. e.g. `alarm_clock_position`` custom`

For "PS" there was `quick _ charge`, the ability to see the current value of the timer.

couriersud, 2020-02-14:
quick_charge was changed on your feedback to screen_info

Wehin, 2020-02-14:
Yes, that command uses exactly the `screen_info` command for` quick_charge` not found

Wehin, 2020-02-16:
i think `--mystic` is not supported by PS, as far as I understand this is LED backlight.

currently V 0.4 output
```
 ./msigd --debug --query --info
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: PS Series
DEBUG: Special 01 b0 : 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d : 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00 
macro_key : '000'
mode : '010'
serial : 'DA1A019340045'
frequency : '030'
unknown190 : '000'
response_time : '001'
alarm_clock : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_index - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on alarm_clock_time - got ''
Try 'msigd --help' for more information.
alarm_position : '002'
screen_assistance : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown271 - got ''
Try 'msigd --help' for more information.
screen_size : '000'
pro_mode : '000'
image_enhancement : '003'
brightness : '073'
contrast : '050'
sharpness : '000'
color_preset : '002'
color_rgb : '���'
gray_level : '009'
low_blue_light : '000'
local_dimming : '000'
hue_rgb : 'bbb'
hue_cmy : 'bbb'
zoom : '000'
zoom_location : '004'
saturation_rgb : 'bbb'
saturation_cmy : 'bbb'
gamma : '003'
input : '000'
pip : '000'
pip_input : '002'
pip_size : '000'
pip_position : '000'
pip_sound_source : '002'
pbp_input1 : '000'
pbp_input2 : '002'
pbp_input3 : '000'
pbp_input4 : '001'
pbp_sound_source : '002'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
audio_source : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown860 - got ''
Try 'msigd --help' for more information.
navi_up : '003'
navi_down : '004'
navi_left : '002'
navi_right : '005'
```

couriersud, 2020-02-16:
All unknown removed from PS series.

couriersud, 2020-04-19:
```
  -w, --wait                 SETTING=VALUE. Wait for SETTING to become
                               VALUE, e.g. macro_key=pressed
```
All issues addressed here should be fixed now.
@Wehin: OK to close?

Wehin, 2020-04-23:
only quick charge settings not found

Wehin, 2020-04-23:
msigd 0.6
```
 ./msigd --debug --query --info
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: PS Series
s140:           <00?>
s150:           <V06>
DEBUG: Special 01 b0: 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 41 00 19 b2 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '012'
serial : 'DA1A019340045'
frequency : '060'
unknown190 : '000'
response_time : '001'
alarm_clock : '000'
alarm_position : '002'
screen_assistance : '000'
screen_size : '000'
pro_mode : '002'
image_enhancement : '003'
brightness : '100'
contrast : '050'
sharpness : '002'
color_preset : '002'
color_rgb : '0x949494'
gray_level : '010'
low_blue_light : '000'
local_dimming : '000'
hue_rgb : 'bbb'
hue_cmy : 'bbb'
zoom : '000'
zoom_location : '004'
saturation_rgb : 'bbb'
saturation_cmy : 'bbb'
gamma : '002'
input : '002'
pip : '000'
pip_input : '003'
pip_size : '000'
pip_position : '000'
pip_sound_source : '002'
pbp_input1 : '002'
pbp_input2 : '003'
pbp_input3 : '002'
pbp_input4 : '002'
pbp_sound_source : '002'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
audio_source : '001'
navi_up : '003'
navi_down : '004'
navi_left : '002'
navi_right : '005'
```

couriersud, 2020-04-23:
I have added 0860 as quick charge setting for PS series.

Wehin, 2020-04-23:
```
./msigd -f quick_charge --debug
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error querying device on quick_charge - got ''
Try 'msigd --help' for more information.
```

couriersud, 2020-04-23:
I am running out of ideas :-(
a) you may try setting the parameter: `msigd --quick_charge on` and check if it changes in the OSD.
b) Is the parameter supported in the Windows "OSD Creator app"? If it is not, than there is not a lot of hope left.

couriersud, 2020-04-23:
I have changed the quick_charge parameter to a write only parameter. Can you please check this works? If not, I am going to remove it completely.

Wehin, 2020-04-24:
not works, sad :(
```
./msigd --quick_charge off --debug
DEBUG: Initializing HID lib
DEBUG: Setting quick_charge to 000
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --quick_charge
Try 'msigd --help' for more information.
```

```
./msigd --quick_charge on --debug
DEBUG: Initializing HID lib
DEBUG: Setting quick_charge to 001
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --quick_charge
Try 'msigd --help' for more information.
```

couriersud, 2020-04-24:
:-( 
I just added some code starting at line 556 in msigd:

```
	// The following are for experimental purposes
	new setting_t(UNKNOWN,	               "00700", "unknown700", 0, 100),
	new setting_t(UNKNOWN,	               "00710", "unknown710", 0, 100),
	new setting_t(UNKNOWN,	               "00720", "unknown720", 0, 100),
	new setting_t(UNKNOWN,	               "00730", "unknown730", 0, 100),
	new setting_t(UNKNOWN,	               "00740", "unknown740", 0, 100),
	new setting_t(UNKNOWN,	               "00870", "unknown870", 0, 100),
	new setting_t(UNKNOWN,	               "00880", "unknown880", 0, 100),
	new setting_t(UNKNOWN,	               "00890", "unknown890", 0, 100),
	new setting_t(UNKNOWN,	               "008:0", "unknown8:0", 0, 100),
```
You can replace UNKNOWN with PS. Then these settings will be compiled into the source and you can check if they are enabled. CAUTION: This is just blindly trying to find the missing setting.

Wehin, 2020-04-24:
try get all unknown, also try set it - no result.
```
./msigd --query --debug | grep unknown
msigd: Error querying device on unknown160 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown1@0 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown271 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown435 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown450 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown470 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown700 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown710 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown720 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown730 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown740 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown870 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown880 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown890 - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on unknown8:0 - got ''
Try 'msigd --help' for more information.
unknown190 : '000'
```

couriersud, 2020-04-24:
Looks like unknown160 has triggered a timeout error. After a timeout error the monitor needs some seconds to react again. In the worst case you have to do a plug/unplug power cycle.
The unknown settings have to be tested individually.
Query:
`msigd --debug -f unknown280`
Setting
`msigd --debug --unknown280 0`
I know it's tedious :-(

Wehin, 2020-04-24:
Looks like 'grep', PS is not freezed. A little later, I still try to get/set them one at a time.

Log without 'grep'.
```
./msigd --query --debug
DEBUG: Initializing HID lib
macro_key : '001'
mode : '012'
serial : 'DA1A019340045'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown160 - got ''
Try 'msigd --help' for more information.
frequency : '060'
unknown190 : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown1@0 - got ''
Try 'msigd --help' for more information.
response_time : '001'
alarm_clock : '000'
alarm_position : '002'
screen_assistance : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown271 - got ''
Try 'msigd --help' for more information.
screen_size : '000'
pro_mode : '002'
image_enhancement : '003'
brightness : '100'
contrast : '050'
sharpness : '002'
color_preset : '002'
color_rgb : '0x949494'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown435 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown450 - got ''
Try 'msigd --help' for more information.
gray_level : '010'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown470 - got ''
Try 'msigd --help' for more information.
low_blue_light : '000'
local_dimming : '000'
hue_rgb : 'bbb'
hue_cmy : 'bbb'
zoom : '000'
zoom_location : '004'
saturation_rgb : 'bbb'
saturation_cmy : 'bbb'
gamma : '002'
input : '002'
pip : '000'
pip_input : '003'
pip_size : '000'
pip_position : '000'
pip_sound_source : '002'
pbp_input1 : '002'
pbp_input2 : '003'
pbp_input3 : '002'
pbp_input4 : '002'
pbp_sound_source : '002'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
audio_source : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown700 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown710 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown720 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown730 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown740 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown870 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown880 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown890 - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown8:0 - got ''
Try 'msigd --help' for more information.
navi_up : '003'
navi_down : '004'
navi_left : '002'
navi_right : '005'
```

Wehin, 2020-04-24:
maybe write test function for all unknown?

do
 - read next `unknown` value
 - read `serial` for check timeout error currently is not triggered
loop

couriersud, 2020-04-24:
Here's my last idea. unknown190 is supported.
Please change
`	new setting_t(PS,                      "00190", "unknown190"),  // returns 56006 on MAG, 000 on PS`
to
`	new setting_t(PS,                      "00190", "unknown190", 0, 100),  // returns 56006 on MAG, 000 on PS`
Try
`./msigd --debug --unknown190 1`

Wehin, 2020-04-24:
yep, i try it first, no result.
```
./msigd --debug -f unknown190;./msigd --debug --unknown190 1
DEBUG: Initializing HID lib
unknown190 : '000'
DEBUG: Initializing HID lib
DEBUG: Setting unknown190 to 001
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --unknown190
Try 'msigd --help' for more information.
```

couriersud, 2020-04-24:
Is the quick charge setting supported in the Windows OSD Creator App?

Wehin, 2020-04-24:
i dunno, no have windows.

couriersud, 2020-04-24:
The information is critical. If it is not supported there, there is practically zero chance we find the setting.

Wehin, 2020-04-24:
```sh
#!/bin/sh
#----------------------------------------------------------------------
# Description: Check list unknown values
# Author: John Wehin
# Created at: Fri Apr 24 16:53:13 SAMT 2020
#----------------------------------------------------------------------
#
declare -a StringArray=("unknown160" "unknown190" "unknown1@0" \
"unknown210" "unknown271" "unknown280" \
"unknown435" "unknown440" "unknown450" "unknown470" \
"unknown700" "unknown710" "unknown720" "unknown730" "unknown740" \
"unknown860" "unknown870" "unknown880" "unknown890" "unknown8:0")

SERIAL=$(./msigd -f serial)

echo "==== ---- Check GET ---- ===="
for val in "${StringArray[@]}"; do
  echo "--- Check ${val} ---"
  CHECK=$(./msigd -f ${val} 2>1)
  CHECK_CODE=$?
  SERIAL_CHECK=$(./msigd -f serial)
  if [ "$SERIAL" =  "$SERIAL_CHECK" ]; then
    echo "TIMEOUT: OK   -- RESULT: $CHECK_CODE"
  else
    echo "TIMEOUT: FAIL -- RESULT: $CHECK_CODE"
    exit 1
  fi
  echo -e "$CHECK"
  echo "-------------------"
done

echo "==== ---- Check SET ---- ===="
for val in "${StringArray[@]}"; do
  echo "--- Check ${val} ---"
  CHECK=$(./msigd --${val} 0 2>1)
  CHECK_CODE=$?
  SERIAL_CHECK=$(./msigd -f serial)
  if [ "$SERIAL" =  "$SERIAL_CHECK" ]; then
    echo "TIMEOUT: OK   -- RESULT: $CHECK_CODE"
  else
    echo "TIMEOUT: FAIL -- RESULT: $CHECK_CODE"
  fi
  echo -e "$CHECK"
  echo "-------------------"
done
```

Wehin, 2020-04-24:
i found how get state :-)
```
./msigd -f unknown180
unknown180 : 1
```
try set
```
./msigd --unknown180 0 --debug
DEBUG: Initializing HID lib
DEBUG: Setting unknown180 to 000
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --unknown180
Try 'msigd --help' for more information.
```

also get/set without error values 0 or 1
```
./msigd -f unknown310 --debug
DEBUG: Initializing HID lib
unknown310 : '000'
```

```
./msigd --unknown310 1 --debug
DEBUG: Initializing HID lib
DEBUG: Setting unknown310 to 001
```

```
/msigd --unknown310 2 --debug
DEBUG: Initializing HID lib
DEBUG: Setting unknown310 to 002
DEBUG: Error 0 reading from HID device
DEBUG: Got unexpected return <>
msigd: Error setting --unknown310
Try 'msigd --help' for more information.
```

couriersud, 2020-04-24:
so unknow310 is quick_charge?

couriersud, 2020-04-24:
On the MAG series `00310` is the eye saver setting. Will this setting make the colours more yellow?

Wehin, 2020-04-24:
`00180` return status of quick charge, dunno how turn it.
can read and set `00310` without visible effect, maybe it work on some `mode`.

couriersud, 2020-04-24:
On the MAG321 i can read and set unknown210 which on the MAG271 and MAG241 is the black tuner setting - no visible effect or changes to the OSD. Looks like unknown310 (eye saver on MAG) behaves like this on the PS series.
I will implement 00180 as read only. Thanks a lot for your efforts and contributions!

The only way understand how to set quick charge would be a Win 10 in a virtualbox. This allows to use wireshark to log the usb communication between the OSD windows app and the monitor.

Wehin, 2020-04-25:
`unknown310` work like eye_saver, not work on some modes:
mode `eyecare` always == `on`
modes `adobe_rgb`, `dci_p3`, `srgb`, `dicom`  == `off`

`unknown310` work with modes `user`, `hdr`, `cinema`, `reader`, `bw`_(ROFL)_

couriersud, 2020-04-25:
Updated code and documentation. Thanks a lot!

couriersud, 2020-04-25:
Closing this now. Adding script provided has been marked in a separate issue.

closed 2020-04-25 by couriersud