couriersud/msigd #6: Tips and Tricks Prestige PS341WU

Wehin, 2020-02-17:
### PS341WU

Param | Value
------------ | -------------
Manufacturer ID | MSI 
Product Code | 3DA1
Physical size | 795.648mm  x 335.664mm
Resolution | 5120 x 2160
Aspect ratio | 64:27
Pixel type | square
Pixel pitch | 0.1554 x 0.1554
Pixels Per Inch (PPI) | 163 (?)
DPMS levels | Standby Suspend Off
Timings supported | GTF, _(CVT?)_
Video ports | 2xHDMI 2.0, 1xDP 1.4, 1xType C (Alternate Mode DP)

**PIP**

*Need some method to get the parameters more accurately

Preset | **Small** _(35%)_
------------ | -------------
Physical size | 277,8552mm x 117,1716mm
Resolution | 1788x754
Aspect ratio | 64:27

Preset | **Medium** _(40%)_
------------ | -------------
Physical size | 318mm x 134mm
Resolution | 1788x754
Aspect ratio | 64:27

Preset | **Large** _(45%)_
------------ | -------------
Physical size | 358mm x 151mm
Resolution | 1788x754
Aspect ratio | 64:27

**PBP**

### Hack and Tricks for
PS341WU have totally broken EDID Modes for use PIP and PBP

### EDID Data

Param | Value
------------ | -------------
**Chromaticity Coordinates**
Red X| 0.680
Red Y| 0.313
Green X| 0.268
Green Y| 0.685
Blue X| 0.151
Blue Y| 0.055
White X| 0.313
White Y| 0.329 
**HDMI**
Display type |  RGB 4:4:4 + YCrCb 4:4:4
 **HDMI 1.4** 
Extraction InfoFrames |  ACP, ISRC1, ISRC2
Supports | 30/36/48 bits/pixel (10/12/16 bits/color)
Supports | YCbCr 4:4:4 Deep Color
Max TMDS Clock Rate| 340 MHz 
**HDMI** | 2.0
EDID version | 1.3
Max TMDS clock | 340MHz
Max dotclock (aka bandwidth) | 600 MHz
Horizontal Sync  | 30-220 kHz
Vertical Refresh  | 48-144 Hz
Color formats | YcBcR 4:4:4  YcBcR 4:2:2 
**DP** | 1.4
EDID version | 1.4
Max dotclock (aka bandwidth) | 760 MHz
Horizontal Sync | 30-135 kHz
Vertical Refresh | 48-61 Hz
Color formats | RGB 4:4:4 YCrCb 4:4:4 YCrCb 4:2:2

**DTD**

Param | Value
------------ | -------------
Pixel Clock | 594MHz
Horizontal Active | 3840
Horizontal Blanking | 560
Vertical Active | 2160
Vertical Blanking | 90
Horizontal Sync Offset | 176
Horizontal Sync Pulse | 88
Vertical Sync Offset | 8
Vertical Sync Pulse | 10
Horizontal Display Size | 800
Vertical Display Size | 330
Horizontal Border | 0
Vertical Border | 0
Interlaced | false
Stereo Mode | 0
Sync Type | 3 

**Common Modes**
640×480 @ 60 Hz
640×480 @ 67 Hz
640×480 @ 72 Hz
640×480 @ 75 Hz
720×400 @ 70 Hz
720×400 @ 88 Hz
800×600  @ 56 Hz
800×600 @ 60 Hz
800×600 @ 72 Hz
800×600 @ 75 Hz
832×624 @ 75 Hz
1024×768 @ 60 Hz
1024×768 @ 72 Hz
1024×768 @ 75 Hz
1024×768i @ 87 Hz
1152x864 @ 60Hz
1152x870 @ 75 Hz
1280x720 @ 60Hz
1280x800 @ 60Hz
1280x960 @ 60Hz
1280x1024 @ 60Hz
1280×1024 @ 75 Hz
1600x1000 @ 60Hz
1680x1050 @ 60Hz
1920x1080 @ 60Hz

**HDMI**
3840x2160@24Hz
3840x2160@25Hz
3840x2160@30Hz

Zackhardtoname, 2020-06-16:
Not sure if it's the right place to ask, but the monitor brightness is weird in that even with a value of 0 (out of 100) it is still sometimes too bright for me... For my other monitor, if I turn the brightness to 0, it would practically be pitch black. Unsure if `msigd` could turn it darker than the value of 0.

Wehin, 2020-06-16:
you can use xrandr for it.
example :
`xrandr --output DisplayPort-0 --brightness 0.3`