couriersud/msigd #7: MAG272QP parameters

elric1789, 2020-03-30:
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG Series
DEBUG: Special 01 b0 : 01 5a 40 00 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 71 00 2f 41 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d : 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00 
macro_key : '000'
mode : '000'
serial : '             '
frequency : '165'
game_mode : '000'
unknown210 : '009'
response_time : '002'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
unknown280 : '001'
zero_latency : '001'
screen_size : '000'
night_vision : '000'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '070'
contrast : '070'
sharpness : '000'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
unknown435 : '001'
input : '002'
pip : '000'
pip_input : '002'
pbp_input : '002'
pip_size : '002'
pip_position : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'

elric1789, 2020-03-30:
OS: Microsoft Windows [Version 10.0.18363.720]
Works for gaming mode switching, input switching etc.

couriersud, 2020-03-30:
Hi Eric,
thanks a lot for supporting the project. According to the MAG272 manual the monitor does not support pip/pbp. 
**Warning: You may have to completely power off your monitor if this fails.**
Can you try enabling pip mode and post the response here?
`msgid --pip pip`

elric1789, 2020-03-30:
I am sure that pip/pbp is not supported. How do i run msgid? I don't have any such (windows) utility as far as I know.

couriersud, 2020-03-30:
Sorry, correct is
`msigd --pip pip`

elric1789, 2020-03-30:
```
$ ./msigd.exe --pip pip
Unexpected id combination <00O><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
msigd: Error setting --pip
Try 'msigd --help' for more information.

```

couriersud, 2020-03-30:
Thanks a lot! Much appreciated!

elric1789, 2020-03-30:
Thank _you_! I was about to return this monitor specifically due to not being able to programmatically switch between input sources. This utility saved me a lot of trouble ;)

couriersud, 2020-04-04:
Added formal support for MAG272 series. Monitors should now be detected and fully supported.

closed 2020-04-04 by couriersud