couriersud/msigd #8: Support for MSI MAG271CQR

cosmicdan, 2020-04-16:
Howdy,

Love the monitor - dislike the OSD. Would love to see support for this thing.

> The id of your monitor, e.g. MAG321CURV
`MAG271CQR`

> The output of msigd --debug --info --query
```
# ./msigd --debug --info --query
DEBUG: Initializing HID lib
Unexpected id combination <006><V19>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG32 Series
s140:           <006>
s150:           <V19>
DEBUG: Special 01 b0 : 01 5a 1a 01 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 71 00 2d 38 00 00 00 00 00 00 00 00 00 00 
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d : 0
macro_key : '000'
mode : '000'
serial : 'FA7T07946004D'
frequency : '060'
game_mode : '000'
unknown210 : '009'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '001'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
unknown280 : '001'
zero_latency : '001'
screen_size : '003'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on night_vision - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on sharpness - got ''
Try 'msigd --help' for more information.
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '���'
unknown435 : '000'
input : '001'
pip : '000'
pip_input : '000'
pbp_input : '000'
pip_size : '002'
pip_position : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '002'
navi_down : '005'
navi_left : '004'
navi_right : '005'

```

> For linux the output of lsusb
``` 
# lsusb
Bus 002 Device 004: ID 0bc2:231a Seagate RSS LLC Expansion Portable
Bus 002 Device 003: ID 174c:5136 ASMedia Technology Inc. ASM1053 SATA 3Gb/s bridge
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 10f5:0292 Turtle Beach Turtle Beach Z22 Chat
Bus 001 Device 011: ID 1462:3fa4 Micro Star International USB 2.0 Hub
Bus 001 Device 010: ID 046d:c332 Logitech, Inc. G502 Proteus Spectrum Optical Mouse
Bus 001 Device 009: ID 1b1c:1b36 Corsair External HDD    
Bus 001 Device 008: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 007: ID 1a86:8072 QinHeng Electronics 
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 006: ID 0bda:0151 Realtek Semiconductor Corp. Mass Storage Device (Multicard Reader)
Bus 001 Device 002: ID 1058:1003 Western Digital Technologies, Inc. WD Elements Desktop (WDE1UBK)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

> Your operating system and version
```
# $ neofetch --off

cosmicdan@COSMICDAN-HOST 
------------------------ 
OS: Arch Linux x86_64 
Host: MS-7B46 2.0 
Kernel: 5.6.3-arch1-1 
Uptime: 23 mins 
Packages: 1500 (pacman), 6 (flatpak) 
Shell: bash 5.0.16 
Resolution: 2560x1440 
DE: Plasma 
WM: KWin 
WM Theme: Breeze 
Theme: Breeze Dark [Plasma], Breeze-Dark [GTK2/3] 
Icons: Breeze Chameleon Dark [Plasma], Breeze Chameleon Dark [GTK2/3] 
Terminal: konsole 
Terminal Font: Hack 9 
CPU: Intel i5-8400 (6) @ 4.000GHz 
GPU: NVIDIA GeForce GTX 1070 Ti 
GPU: Intel UHD Graphics 630 
Memory: 5624MiB / 32057MiB 
```

> any other information which might be helpful, e.g. OSD setting xyz is not supported.
Seems to support everything listed.

Photo of service menu:
[image]

Am happy to tinker around in the source if it takes a lot of back-and fourth to add support for a new monitor.

Cheers!

couriersud, 2020-04-17:
Thanks a lot for supporting the project!

There were a number of timeout errors in the log starting with night_vision which is not supported on your monitor.

Can you please run the following commands and post the output here? These settings should be supported. By running them one by one we make sure that an error doesn't produce further errors.

```
./msigd --debug -f pro_mode 
./msigd --debug -f eye_saver
./msigd --debug -f image_enhancement
./msigd --debug -f brightness
./msigd --debug -f contrast
./msigd --debug -f sharpness
```

cosmicdan, 2020-04-17:
Ah right, makes sense.

For some reason now I'm just getting an error no matter the command. I restarted machine, rebuilt... haven't changed a thing since :\

```
# ./msigd --debug --info --query
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error on device identification
Try 'msigd --help' for more information.
```

Factory Reset maybe?

couriersud, 2020-04-17:
Try plugging out the power cable of the monitor, wait 5 sec and plug it in again.

couriersud, 2020-04-17:
Hi Daniel,

I now have added preliminary support for the MAG271 series. Can you please check actual git if this working better for you?

cosmicdan, 2020-04-18:
Thanks so much - everything seems to be working (that I checked so far). That same debug output looks all good to me :)

```
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG271 Series
s140:           <006>
s150:           <V19>
DEBUG: Special 01 b0 : 01 5a 1a 01 00 00 00 00 00 00 00 00 00 00 00 00 
DEBUG: Special 01 b4 : 01 5a 71 00 2d 38 00 00 00 00 00 00 00 00 00 00 
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d : 0
macro_key : '000'
serial : 'FA7T07946004D'
frequency : '060'
game_mode : '000'
black_tuner : '00:'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '001'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
zero_latency : '001'
screen_size : '003'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '040'
contrast : '050'
sharpness : '000'
color_preset : '003'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '���'
unknown435 : '000'
input : '001'
pip_input : '000'
pbp_input : '000'
pip_size : '002'
pip_position : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '005'
navi_down : '005'
navi_left : '005'
navi_right : '005'
```

cosmicdan, 2020-04-18:
Dunno what that 'color_rgb' value is about, though... that normal?

couriersud, 2020-04-18:
The debug output dumps raw values. Use `./msigd --query` for meaningful values. 
I'll reopen for now so that will be reminded to use a hex dump for such cases.

cosmicdan, 2020-04-19:
Ah right, that makes sense. Not an important option but wanted to make sure.

Set up some good shortcuts to switch inputs, noticed there's no option to switch pip/pbp on this monitor. Not possible or just not had access to a device to support it? I'm wondering if I could dump memory before/after or something like that as it might just be a different offset or something? (I don't know how HID works really).

⁣Get BlueMail for Android ​

Wehin, 2020-04-19:
i think it's not directly mapped to some memory and controlled by chip

couriersud, 2020-04-19:
Fixed the missing pip/pbp option in https://github.com/couriersud/msigd/commit/db6f78d84e96dd2b7f3d1547cd2538ac43b8175c
`
      --pip                  values: off pip pbp 
`

couriersud, 2020-04-19:
Closing this for now since all issues described were fixed. Please reopen if there are other issues.

closed 2020-04-19 by couriersud