couriersud/msigd #9: Bildung on Arch-Linux

mike-vivas, 2020-04-17:
Hi,
thanks a lot! Awesome piece of code to control my new MSI monitor!

Just had a problem compiling it under arch. "Make" did not find the lusb library.
I had to modify the makefile and specify "lusb-1.0" to make it compile.
It might be worth to add "TARGETOS=arch" to the makefile and use this specific libusb version for arch (might reduce problems to other arch users).

By the way, the needed libraries under arch are called: libusb and hidapi
sudo pacman -S libusb hidapi

Thanks for sharing the code!

couriersud, 2020-04-17:
Hi Mike,

I have added the `TARGETOS=arch` option to the makefile and updated README.md.
Can you please check if this is now working?

What monitor are you using? The information would be helpful to update the supported monitor matrix in README.md.

cosmicdan, 2020-04-18:
Just FYI, I didn't have to do anything special on Arch - I just compiled as normal while installing those packages.

With the latest update, I can still just call `make` fine. With TARGETOS=arch I get an error which is fixed with `-lhidapi-libusb` (instead of `-lhidapi`) :)

mike-vivas, 2020-04-19:
Hi @couriersud,
thank you for the changes and sorry for the delay.
The lusb-1.0 is now working but, as cosmicdan mentioned, now the hidapi-library link is broken.
In arch the library is called -lhidapi-hidraw (like for the normal "linux" targetos)
could you please change it back in the makefile?
(The pacman command that you put in the README is correct - the library is installed as "hidapi" but it has to be linked as "lhidapi-hidraw" or "lhidapi-libusb" - I do not really know the difference but it compiles with both)

@cosmicdan: thanks for the feedback.
How is this possible? Before posting I also checked with a friend of mine running on arch (kernel 5.6.4). He also installed the libusb library but was not able to build with the -lusb link. He had to change it to -lusb-1.0 as well.
What does it depend upon? It is just a missing link in /usr/lib/ I guess...

cosmicdan, 2020-04-19:
I believe there is a mention on the readme about using hidraw Vs libusb, iirc.

Not sure how it was working for me beforehand, though I do other development on my machine too so with your new info it's likely some compatibility I've previously implemented (possibly via ArchWiki guide on AOSP compiling, which is typically a Debian-land process).

Good to hear that that minor makefile change works for other Arch too now.

⁣Get BlueMail for Android ​

couriersud, 2020-04-19:
When  `make TARGETOS=arch` is used, LIBS now default to
`LIBS = -lusb-1.0 -lhidapi-hidraw`
@mike-vivas : Can you please check this is working now? Thanks a lot!

mike-vivas, 2020-04-21:
works perfectly! Thanks!

As for your question about the monitor: I've got a MAG321CURV, which seems to be quite well documented in the README...
should you need any additional information, let me know.

closed 2020-04-21 by mike-vivas

couriersud, 2020-04-21:
Mike, thanks a lot!
And well, the MAG321CURV is the monitor in front of me :-)