couriersud/msigd #10: Timeout for input change

mike-vivas, 2020-04-21:
Hi Couriersud,

I have the following problem: 
I switch to an input (say DisplayPort) with msigd and it works.
The input-signal does not come fast enough to the new input (DisplayPort), the monitor timeout runs out and the monitor brings me back to my previous input (say HDMI1).

Now the question: do you know a way to change this timeout when changing input? To make it longer?

The other way would be to provide the input-signal faster - but I fear I cannot do this since it is a virtual machine starting up.

Thanks in advance for any tip!

couriersud, 2020-04-21:
I have the same problem: Source hdmi not providing input
```
./msigd --input hdmi1; sleep 1; ./msigd -f input
input : hdmi1
```
After the monitor switched back (no input on hdmi1)
```
 ./msigd -f input
input : dp
```
The only way would be to wait at least 5 seconds:
```
./msigd --input hdmi1; sleep 5; ./msigd -f input
input : dp
```
and try again - say 3 times. Should be doable in a shell script.

I may do that in msigd. But that's really not not the purpose of the program. 
Something like

```
INPUT=./msigd --input hdmi1; sleep 5; ./msigd -f input | cut -d: -f2 | sed -e "s/ //g"
if [ _$INPUT  = _hmdi1 ]; then echo OK; else echo NOT OK; fi
```

should give you an idea.

mike-vivas, 2020-04-22:
Yes, that is my current solution as well...trying again and again until the input-signal is there.

> I may do that in msigd. But that's really not not the purpose of the program.
Completely agree...I was actually hoping that you came across one parameter which could be set via USB and that would modify the timeout duration :-)
For example unknown210 or so :-)

Either I'll give it a try with the unknown parameters or I'll simple accept the other solution.

Thanks!

couriersud, 2020-04-22:
> Completely agree...I was actually hoping that you came across one parameter which could be set via USB and that would modify the timeout duration :-)

I was hoping as well to discover some undocumented functionality when writing this program. Until now, none was found :-( 
`unknown210` is the "black tuner" setting on other monitors. On the MAG321CURV you can set it to  values between 0 and 9, but I couldn't determine what it actually does.
It looks like the monitors just provide the OSD functionality over USB and nothing more.

It would be nice if you share your solution here. It would be nice to have a contrib folders with examples of what users do with the tool in addition to the example scripts I wrote.

mike-vivas, 2020-04-25:
Hey!
I'll gladly do that...
I tried to create a branch "contrib-mike" on my machine and then push it to this project (so that I can open a "Pull Request" and you can review it before merging it into the master)...
but it seems that I do not have the rights to push the branch...
shall I do this way? Would you like me to share it in another way?

couriersud, 2020-04-25:
You first have to create a fork. Github shows me that you have done so. Afterwards you do the changes on the fork and create a pull request. Details on this (including links how to create the fork) are here:

https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

mike-vivas, 2020-04-25:
ok, was not sure whether I really had to fork the whole project just to add a branch...
but it is done now and the pull request is open

couriersud, 2020-04-25:
Thanks a lot for the contribution! I have merged the pull request. Thus closing this issue.

closed 2020-04-25 by couriersud