couriersud/msigd #11: Unknow monitor MAG241C

intello21, 2020-04-22:
- ## The id of your monitor 
`MAG241C`
- ## The output of msigd --debug --info --query
```
DEBUG: Initializing HID lib
Unexpected id combination <002><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A**********0
Monitor Series: MAG32 Series
s140:           <002>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 15 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 50 20 7e 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
mode : '000'
serial : 'EA**********5'
frequency : '060'
game_mode : '000'
unknown210 : '009'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
unknown280 : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on night_vision - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on contrast - got '56006'
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on sharpness - got ''
Try 'msigd --help' for more information.
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
msigd: Error querying device on pip - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pbp_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_size - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_position - got '56006'
Try 'msigd --help' for more information.
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '002'
navi_down : '003'
navi_left : '001'
navi_right : '005'
```

- ## For linux the output of lsusb
`  Bus 003 Device 006: ID 1462:3fa4 Micro Star International USB2.0 Hub `
- ## Your operating system and version
`Linux 5.4.33-1-MANJARO`
- ## Any other information which might be helpful
I don't know but I can give more information as you need

couriersud, 2020-04-22:
Thanks a lot! Much appreciated!
Can you please try the changes I committed to support this monitor and provide another round of 
`msigd --debug --info --query`

intello21, 2020-04-22:
Thanks for you work
Now I have this result 
```
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error on device identification
Try 'msigd --help' for more information.
```

couriersud, 2020-04-22:
Try unplugging the power from the monitor. 
"Error 0" indicates a timeout error. We have seen this before after the nightvision setting was queried. It is now disabled for the MAG241C.

intello21, 2020-04-22:
Okay. Now I have the same result as in my first post

couriersud, 2020-04-23:
I missed that the MAG241C does not support the zero_latency setting. I have removed it now.
Can you please try again based on latest commit (https://github.com/couriersud/msigd/commit/a371467a3c9f57995b93b40a4eb59c959f13cf99)?

intello21, 2020-04-23:
Same result except for the contrast
```
contrast : '070'
```

couriersud, 2020-04-23:
There was a copy/paste error. Device identification should now work.

intello21, 2020-04-23:
Yes the device identification seem to work fine
```
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A**********0
Monitor Series: MAG241 Series
s140:           <002>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 15 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 50 20 7e 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
serial : 'EA**********5'
frequency : '060'
game_mode : '000'
black_tuner : '009'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on sharpness - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on color_preset - got '56006'
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on color_red - got ''
Try 'msigd --help' for more information.
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '002'
navi_down : '003'
navi_left : '001'
navi_right : '005'
```

couriersud, 2020-04-23:
Ok, we are getting there ... Can you now please try running the following commands 

`./msigd -f brightness --debug`
`./msigd -f pro_mode --debug`
`./msigd -f eyer_saver --debug`
`./msigd -f image_enhancement --debug`
`./msigd -f contrast --debug`
`./msigd -f image_enhancement --debug`
`./msigd -f color_red --debug`
`./msigd -f screen_size --debug`
`./msigd -f color_preset --debug`

These need to be executed individually. A timeout (ERROR 0) seems to block the device for a couple of seconds and thus spoils further reads.

intello21, 2020-04-23:
./msigd -f brightness --debug
```
DEBUG: Initializing HID lib
brightness : '000
```
./msigd -f pro_mode --debug
```
DEBUG: Initializing HID lib
pro_mode : '000'
```
./msigd -f eyer_saver --debug
```
DEBUG: Initializing HID lib
msigd: Unsupported query filter elements: 1 of 1
Try 'msigd --help' for more information.
```
./msigd -f image_enhancement --debug
```
DEBUG: Initializing HID lib
image_enhancement : '000'
```
./msigd -f contrast --debug
```
DEBUG: Initializing HID lib
contrast : '070'
```
./msigd -f image_enhancement --debug
```
DEBUG: Initializing HID lib
image_enhancement : '000'
```
./msigd -f color_red --debug
```
DEBUG: Initializing HID lib
color_red : '100'
```
./msigd -f screen_size --debug
```
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
```
./msigd -f color_preset --debug
```
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error on device identification
Try 'msigd --help' for more information.
```

intello21, 2020-04-23:
When I use `./msigd -f screen_size --debug` I need to disconnect/reconnect power to execute another command without timeout  message

couriersud, 2020-04-23:
Yes, that one seems to be the culprit. I have removed it for now. Please rerun

`msigd --debug --info --query`

and post results here. It seems it is not supported via USB, although the manual documents it as on OSD setting.

intello21, 2020-04-23:
Everything seem to work fine with `msigd --debug --info --query` 
```
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A**********0
Monitor Series: MAG241 Series
s140:           <002>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 15 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 50 20 7e 00 00 00 00 00 00 00 00 00 00
macro_key : '000'
serial : 'EA**********5'
frequency : '060'
game_mode : '000'
black_tuner : '009'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '000'
contrast : '070'
sharpness : '000'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
unknown860 : '001'
navi_up : '002'
navi_down : '003'
navi_left : '001'
navi_right : '005'
```

For the `./msigd -f screen_size --debug` I have this result (and my screen keep connected)
```
DEBUG: Initializing HID lib
msigd: Unsupported query filter elements: 1 of 1
Try 'msigd --help' for more information.
```

Thank again for your work :)

couriersud, 2020-04-23:
Good that it is working now. Will update status from WIP to supported later.
One question, please: Is setting the screen size supported in the Windows OSD app?

couriersud, 2020-04-25:
The screen size issue is described in README.md now. Thus closing this issue.

closed 2020-04-25 by couriersud