couriersud/msigd #14: Unknown monitor - MSI MAG272

gbirchley, 2020-05-12:
Just tried msigd with new MSI MAG272
Running on Linux - Fedora 32
I haven't tried any functions - no quite sure what I'm doing - but will report back when I do!

$ sudo ./msigd --debug --info --query  
DEBUG: Initializing HID lib
Unexpected id combination <00L><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG32 Series
s140:           <00L>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 40 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 71 00 29 70 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
DEBUG: Error -1 writing to HID device
msigd: Error querying device on macro_key - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on serial - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on frequency - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on game_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on unknown210 - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on response_time - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on enable_dynamic - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on hdcr - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on refresh_display - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on refresh_position - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on alarm_clock - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on alarm_clock_index - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on alarm_clock_time - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on alarm_position - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on screen_assistance - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on unknown280 - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on zero_latency - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on night_vision - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on sharpness - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_preset - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_red - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_green - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_blue - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_rgb - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on unknown435 - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on input - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pip - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pip_input - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pbp_input - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pip_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on pip_position - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on osd_language - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on osd_transparency - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on osd_timeout - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on sound_enable - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on rgb_led - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on navi_up - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on navi_down - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on navi_left - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on navi_right - got ''
Try 'msigd --help' for more information.

$ sudo lsusb
Bus 002 Device 006: ID 046d:c218 Logitech, Inc. F510 Gamepad [DirectInput Mode]
Bus 002 Device 009: ID 050d:0017 Belkin Components B8T017 Bluetooth+EDR 2.1 / F4U017 USB 2.0 7-port Hub
Bus 002 Device 008: ID 0a5c:4503 Broadcom Corp. Mouse (Boot Interface Subclass)
Bus 002 Device 007: ID 0a5c:4502 Broadcom Corp. Keyboard (Boot Interface Subclass)
Bus 002 Device 005: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 002 Device 004: ID 046d:c05b Logitech, Inc. M-U0004 810-001317 [B110 Optical USB Mouse]
Bus 002 Device 021: ID 04d9:1203 Holtek Semiconductor, Inc. Keyboard
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 010: ID 1462:3fa4 Micro Star International 
Bus 001 Device 008: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 2304:0225 Pinnacle Systems, Inc. Remote Kit Infrared Transceiver
Bus 003 Device 002: ID 046d:c218 Logitech, Inc. F510 Gamepad [DirectInput Mode]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ uname -a
Linux media.weaver 5.6.8-300.fc32.x86_64 #1 SMP Wed Apr 29 19:01:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

gbirchley, 2020-05-12:
Quick update - I have altered the following settings are recognised:

./msigd --brightness 60
./msigd --pro_mode cinema
./msigd --night_vision off
./msigd --navi_up

While the only one that is not is:

$ sudo ./msigd --mode user
Unexpected id combination <00L><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
msigd: Error setting --mode
Try 'msigd --help' for more information.

Many thanks for some useful software :)

Wehin, 2020-05-12:
```
su-c "rpm -qi libusb-devel hidapi-devel"
```

couriersud, 2020-05-13:
@gbirchley Thanks a lot for submitting the details. I will add the monitor later to the list of supported monitors and review the MAG272 manual for necessary changes.

couriersud, 2020-05-13:
Added the monitor to the list. Can you please try again and post the output of
`msigd --info --debug --query` again?

Which of the following 272 monitors do you own?
 MAG272
 MAG272C
 MAG272CR
 MAG272R
 MAG272QP
 MAG272QR
 MAG272CR
 MAG272CQR
 MAG272CRX
 MAG272QP
 MAG272QR

gbirchley, 2020-05-13:
Hi, many thanks for that. The model is a MAG272 (i.e. the first 
on your list)
$ sudo ./msigd --debug --info --query
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG272 Series
s140:           <00L>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 40 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 71 00 29 70 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '012'
serial : '             '
frequency : '060'
game_mode : '000'
unknown210 : '009'
response_time : '001'
enable_dynamic : '000'
hdcr : '001'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
zero_latency : '001'
screen_size : '000'
night_vision : '004'
pro_mode : '002'
eye_saver : '000'
image_enhancement : '001'
brightness : '070'
contrast : '070'
sharpness : '000'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
unknown435 : '001'
input : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
msigd: Error querying device on screen_info - got '56006'
Try 'msigd --help' for more information.
sound_enable : '001'
rgb_led : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'

@Wehin -
$ su -c "rpm -qi libusb-devel hidapi-devel"
Password: 
Name        : libusb-devel
Epoch       : 1
Version     : 0.1.5
Release     : 16.fc32
Architecture: x86_64
Install Date: Tue 12 May 2020 21:44:33 BST
Group       : Unspecified
Size        : 10330
License     : LGPLv2+
Signature   : RSA/SHA256, Thu 30 Jan 2020 05:10:09 GMT, Key ID 6c13026d12c944d0
Source RPM  : libusb-0.1.5-16.fc32.src.rpm
Build Date  : Wed 29 Jan 2020 11:47:36 GMT
Build Host  : buildhw-12.phx2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : http://sourceforge.net/projects/libusb/
Bug URL     : https://bugz.fedoraproject.org/libusb
Summary     : Development files for libusb
Description :
This package contains the header files, libraries and documentation needed to
develop applications that use libusb-0.1. However new applications should use
libusb-1.0 library instead of this one.
Name        : hidapi-devel
Version     : 0.9.0
Release     : 3.fc32
Architecture: x86_64
Install Date: Tue 12 May 2020 21:44:33 BST
Group       : Unspecified
Size        : 14508
License     : GPLv3 or BSD
Signature   : RSA/SHA256, Wed 29 Jan 2020 18:44:48 GMT, Key ID 6c13026d12c944d0
Source RPM  : hidapi-0.9.0-3.fc32.src.rpm
Build Date  : Wed 29 Jan 2020 04:24:19 GMT
Build Host  : buildhw-08.phx2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : https://github.com/libusb/hidapi
Bug URL     : https://bugz.fedoraproject.org/hidapi
Summary     : Development files for hidapi
Description :
This package contains development files for hidapi which provides access to
USB and Bluetooth HID-class devices.

couriersud, 2020-05-13:
@gbirchley : looks good :-) Thanks a lot for the feedback.
Only setting not supported is screen-info. I'll fix that.

Wehin, 2020-05-14:
hmm...  unknown435 : '001' , can you find this option in osd?

couriersud, 2020-05-14:
435 seems to be a color setting and read-only on all MAG series.
There is also unknown440 which seems to be write-only.
I  couldn't figure out what these do.

couriersud, 2020-06-24:
Everything fixed. Closing.

closed 2020-06-24 by couriersud