couriersud/msigd #15: How to run on Manjaro

Zackhardtoname, 2020-06-11:
Sorry for such a noob question.
I am using Manjaro 20.01.

I ran 
```
sudo pacman -S libusb hidapi
make TARGETOS=arch
```
The output suggests it's successfully installed

> ➜  msigd git:(master) sudo pacman -S libusb hidapi
> make TARGETOS=arch
> [sudo] password for zack: 
> warning: libusb-1.0.23-2 is up to date -- reinstalling
> warning: hidapi-0.9.0-1 is up to date -- reinstalling
> resolving dependencies...
> looking for conflicting packages...
> 
> Packages (2) hidapi-0.9.0-1  libusb-1.0.23-2
> 
> Total Installed Size:  0.26 MiB
> Net Upgrade Size:      0.00 MiB
> 
> :: Proceed with installation? [Y/n] y
> (2/2) checking keys in keyring                       [############################] 100%
> (2/2) checking package integrity                     [############################] 100%
> (2/2) loading package files                          [############################] 100%
> (2/2) checking for file conflicts                    [############################] 100%
> (2/2) checking available disk space                  [############################] 100%
> :: Processing package changes...
> (1/2) reinstalling libusb                            [############################] 100%
> (2/2) reinstalling hidapi                            [############################] 100%
> :: Running post-transaction hooks...
> (1/1) Arming ConditionNeedsUpdate...
> creating .depend
> Compiling src/msigd.cpp...
> In file included from src/msigd.cpp:20:
> src/phid.h: In member function ‘int usbdev_t::control_msg_write(int, int, int, int, void*, int, int)’:
> src/phid.h:62:30: warning: unused parameter ‘timeout’ [-Wunused-parameter]
>    62 |   void *bytes, int size, int timeout)
>       |                          ~~~~^~~~~~~
> src/phid.h: In member function ‘int usbdev_t::control_msg_read(int, int, int, int, void*, int, int)’:
> src/phid.h:81:27: warning: unused parameter ‘requesttype’ [-Wunused-parameter]
>    81 |  int control_msg_read(int requesttype, int request, int value, int index,
>       |                       ~~~~^~~~~~~~~~~
> src/phid.h:81:44: warning: unused parameter ‘request’ [-Wunused-parameter]
>    81 |  int control_msg_read(int requesttype, int request, int value, int index,
>       |                                        ~~~~^~~~~~~
> src/phid.h:81:57: warning: unused parameter ‘value’ [-Wunused-parameter]
>    81 |  int control_msg_read(int requesttype, int request, int value, int index,
>       |                                                     ~~~~^~~~~
> src/phid.h:81:68: warning: unused parameter ‘index’ [-Wunused-parameter]
>    81 |  int control_msg_read(int requesttype, int request, int value, int index,
>       |                                                                ~~~~^~~~~
> src/phid.h:82:9: warning: unused parameter ‘bytes’ [-Wunused-parameter]
>    82 |   void *bytes, int size, int timeout)
>       |   ~~~~~~^~~~~
> src/phid.h:82:20: warning: unused parameter ‘size’ [-Wunused-parameter]
>    82 |   void *bytes, int size, int timeout)
>       |                ~~~~^~~~
> src/phid.h:82:30: warning: unused parameter ‘timeout’ [-Wunused-parameter]
>    82 |   void *bytes, int size, int timeout)
>       |                          ~~~~^~~~~~~
> src/phid.h: In member function ‘int usbdev_t::read(void*, unsigned int, unsigned int)’:
> src/phid.h:155:15: warning: comparison of integer expressions of different signedness: ‘int’ and ‘unsigned int’ [-Wsign-compare]
>   155 |   if (result  < bs)
>       |       ~~~~~~~~^~~~
> src/msigd.cpp: In member function ‘virtual std::string alarm4x_t::decode(std::string)’:
> src/msigd.cpp:437:33: warning: unused parameter ‘val’ [-Wunused-parameter]
>   437 |  std::string decode(std::string val) override
>       |                     ~~~~~~~~~~~~^~~
> Linking msigd...
> g++ -o msigd  src/msigd.o -lusb-1.0 -lhidapi-hidraw

However, when I have
```
➜  ~ msigd             
zsh: command not found: msigd
```
Not working in bash neither. Could you please kindly help? Thank you!

couriersud, 2020-06-11:
try 
`./msigd`
Your current directory is not part of your path (for good reasons).

Zackhardtoname, 2020-06-12:
Resolved... Sorry for such a dumb question... Works great so far!

couriersud, 2020-06-13:
Good to hear!
May I ask which monitor - so that I can update the supported monitor list?
The output of `./msigd --debug --info --query` also would be helpful.

Zackhardtoname, 2020-06-15:
I use this on MSI prestige ps341wu while my PC is also connected to Dell P2715Q (obviously only works for the former, and looks like there are general monitor controlling software I could use for the latter).

```
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: PS Series
s140:           <00?>
s150:           <V06>
DEBUG: Special 01 b0: 01 5a 37 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 41 00 19 d2 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '010'
serial : 'DA1A129280018'
frequency : '060'
quick_charge : '000'
unknown190 : '000'
response_time : '002'
alarm_clock : '000'
alarm_position : '000'
screen_assistance : '000'
screen_size : '000'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '005'
contrast : '005'
sharpness : '000'
color_preset : '002'
color_rgb : 'bbb'
gray_level : '010'
low_blue_light : '000'
local_dimming : '000'
hue_rgb : 'bbb'
hue_cmy : 'bbb'
zoom : '000'
zoom_location : '000'
saturation_rgb : 'bbb'
saturation_cmy : 'bbb'
gamma : '002'
input : '002'
pip : '000'
pip_input : '000'
pip_size : '001'
pip_position : '000'
pip_sound_source : '000'
pbp_input1 : '002'
pbp_input2 : '000'
pbp_input3 : '000'
pbp_input4 : '000'
pbp_sound_source : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
audio_source : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'
```

couriersud, 2020-06-15:
Thanks a lot!
Closing this now.

closed 2020-06-15 by couriersud