couriersud/msigd #17: Please support Optix MAG272CQR

andaag, 2020-07-04:
Thanks a lot for the tool! If there's anything else I can do to help support this let me know.

```
DEBUG: Initializing HID lib
Unexpected id combination <00E><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG32 Series
s140:           <00E>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 33 01 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 61 00 2a 84 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '001'
mode : '000'
serial : 'CA6T0501700GL'
frequency : '165'
game_mode : '000'
unknown210 : '009'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
unknown280 : '001'
zero_latency : '001'
screen_size : '000'
night_vision : '004'
pro_mode : '002'
eye_saver : '000'
image_enhancement : '000'
brightness : '063'
contrast : '072'
sharpness : '000'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '001'
input : '002'
pip : '000'
pip_input : '002'
pbp_input : '002'
pip_size : '002'
pip_position : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'
```

couriersud, 2020-07-04:
Thanks a lot! Added the monitor to the MAG272 family. Can you please rerun

`msigd --info --debug --query` 

and post the output here?

couriersud, 2020-07-04:
Looking at the output I notice that the picture in picture setting seem to be supported. The manual doesn't mention picture in picture.

Before recompiling, can you please try `./msigd --pip pip --pip_input hdmi1` (or some other input)?

andaag, 2020-07-05:
> Looking at the output I notice that the picture in picture setting seem to be supported. The manual doesn't mention picture in picture.
> 
> Before recompiling, can you please try `./msigd --pip pip --pip_input hdmi1` (or some other input)?

This is from commit 3395b28a (before the above patch):
```
~/src/msigd ((3395b28...))$ sudo ./msigd --pip pip --pip_input dp
Unexpected id combination <00E><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
msigd: Error setting --pip
Try 'msigd --help' for more information.
```

andaag, 2020-07-05:
> Thanks a lot! Added the monitor to the MAG272 family. Can you please rerun
> 
> `msigd --info --debug --query`
> 
> and post the output here?

```
 sudo ./msigd --info --debug --query
DEBUG: Initializing HID lib
DEBUG: Error 0 reading from HID device
msigd: Error on icdeve identification
Try 'msigd --help' for more information.
```

Just to check I also built it with USE_HIDAPI=0:
```
sudo ./msigd --info --debug --query
DEBUG: Initializing usb_lib
DEBUG: Scanning USB devices...
DEBUG: Found device MSI Gaming Controller 

DEBUG: Interface 0 alt 0 class 3
DEBUG: Endpoint 01 0040
DEBUG: Endpoint 1 does is not of type output
DEBUG: Endpoint 02 0040
DEBUG: Endpoint 2 does is not of type input
DEBUG: Found 1 interfaces, using interface 0
DEBUG: Setting Configuration
DEBUG: Driver is usbhid
DEBUG: Error -110 reading from USB device
msigd: Error on device identification
Try 'msigd --help' for more information.
DEBUG: Releasing interface 0
```
(note that in order to build this I needed to add #include <cstring> to src/pusb.h)

Thanks so far, anything else I can do to help figure this out?

couriersud, 2020-07-05:
> Anything else I can do to test this?

No, thanks a lot for your support. PIP can be queried, but not set. A bit weird and looks like software feature disabling. But it was worth a try. :-)

couriersud, 2020-07-05:
> (by the way, during testing I tried building with libusb. In order to build this I needed to add #include to src/pusb.h)

Thanks a lot, I have fixed this. Please let me know if there are settings in the OSD which are not supported by the misgd tool. If this is the case, I'll leave the issue open until we resolved those.

andaag, 2020-07-07:
I had a quick look at the osd tool in windows, and it looks like it's all there.

Thanks a lot 👍

closed 2020-07-07 by andaag