couriersud/msigd #18: Unknown Monitor: MAG321CQR

Deathof1, 2020-08-17:
Operating system: 5.6.19-2-MANJARO #1 SMP

Monitor: MSI Optix MAG321CQR

contents of ./msigd --info --debug --query

DEBUG: Initializing HID lib
Unexpected id combination <00:><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG32 Series
s140:           <00:>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 23 03 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 e7 07 51 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
mode : '004'
serial : 'KA3H019040486'
frequency : '144'
game_mode : '004'
unknown210 : '009'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '001'
screen_assistance : '000'
unknown280 : '001'
zero_latency : '001'
screen_size : '003'
msigd: Error querying device on night_vision - got '56006'
Try 'msigd --help' for more information.
pro_mode : '001'
eye_saver : '000'
image_enhancement : '001'
brightness : '080'
contrast : '060'
sharpness : '002'
color_preset : '000'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
pip : '000'
pip_input : '002'
pbp_input : '002'
pip_size : '000'
pip_position : '000'
osd_language : '001'
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'

contents of lsusb

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c08b Logitech, Inc. G502 HERO Gaming Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1b1c:1c0b Corsair                       
Bus 001 Device 006: ID 0b0e:0305 GN Netcom Jabra EVOLVE LINK MS
Bus 001 Device 015: ID 1004:633e LG Electronics, Inc. G2/G3 Android Phone [MTP/PTP/Download mode]
Bus 001 Device 003: ID 1b1c:1b3d Corsair Corsair Gaming K55 RGB Keyboard
Bus 001 Device 008: ID 2516:0051 Cooler Master Co., Ltd. AMD SR4 lamplight Control
Bus 001 Device 004: ID 1462:3fa4 Micro Star International LGMS550
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

couriersud, 2020-08-18:
Add preliminary support for this monitor.
Can you please run `./msigd --info --debug --query` again and post results here?

Deathof1, 2020-08-18:
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG321CQR
s140:           <00:>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 23 03 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 e7 07 51 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
mode : '000'
serial : 'KA3H019040486'
frequency : '144'
game_mode : '000'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
zero_latency : '001'
screen_size : '003'
pro_mode : '001'
eye_saver : '000'
image_enhancement : '000'
brightness : '080'
contrast : '050'
sharpness : '000'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
pip : '000'
pip_input : '002'
pbp_input : '002'
pip_size : '000'
pip_position : '000'
osd_language : '001'
osd_transparency : '003'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '001'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'

couriersud, 2020-08-18:
Cool - looks right. 
Closing this one. If you run into another issue, please open a new one.

closed 2020-08-18 by couriersud