couriersud/msigd #20: Unknown Monitor MSI Optix MAG241CR

Skyyblaze, 2020-09-11:
The MSIGD tool works fine as far as I can tell but my MSI Optix MAG241CR still reports as an unknown monitor.

`DEBUG: Initializing HID lib
Unexpected id combination <004><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG321CQR
s140:           <004>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 19 01 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 ea 06 55 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
mode : '000'
serial : 'EA2T308300377'
frequency : '144'
game_mode : '000'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
msigd: Error querying device on sharpness - got '56006'
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on color_preset - got ''
Try 'msigd --help' for more information.
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
msigd: Error querying device on pip - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pbp_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_size - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_position - got '56006'
Try 'msigd --help' for more information.
osd_language : '001'
osd_transparency : '002'
osd_timeout : '030'
sound_enable : '001'
rgb_led : '000'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'`

couriersud, 2020-09-11:
Thanks for providing the details!
Can you please update to latest commit and retry? 
Please post the output of  `msigd --info --debug --query` again.

Skyyblaze, 2020-09-12:
> 
> 
> Thanks for providing the details!
> Can you please update to latest commit and retry?
> Please post the output of `msigd --info --debug --query` again.

Sure thing, thanks for the fast fix! Looking good as far as I can tell:

```
DEBUG: Initializing HID lib
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG241 Series
s140:           <004>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 19 01 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 ea 06 55 00 00 00 00 00 00 00 00 00 00
macro_key : '000'
serial : 'EA2T308300377'
frequency : '144'
game_mode : '000'
black_tuner : '009'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '080'
contrast : '080'
sharpness : '000'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
osd_language : '001'
osd_transparency : '002'
osd_timeout : '030'
sound_enable : '001'
rgb_led : '000'
navi_up : '002'
navi_down : '003'
navi_left : '004'
navi_right : '005'
```

closed 2021-05-02 by couriersud