couriersud/msigd #22: Unknown monitor MPG341CQR

glaon, 2020-11-22:
Hi, the tool works fine for most options, great work! I was able build very basic argos shell extension in ubuntu to alter settings.

Not working currently as far as I can tell:
`pro_mode` with `HDR`
`mystic`
`night_vision` 

```
DEBUG: Initializing HID lib
Unexpected id combination <00>><V09>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: MAG321CQR
s140:           <00>>
s150:           <V09>
DEBUG: Special 01 b0: 01 5a 36 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 61 00 39 fa 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '011'
serial : '_____________'
frequency : '144'
msigd: Error querying device on game_mode - got '5b00000000'
Try 'msigd --help' for more information.
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '_00'
alarm_position : '000'
screen_assistance : '__0'
free_sync : '001'
zero_latency : '001'
screen_size : '000'
pro_mode : '001'
eye_saver : '001'
image_enhancement : '000'
brightness : '060'
contrast : '060'
sharpness : '003'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on unknown435 - got ''
Try 'msigd --help' for more information.
input : '002'
pip : '000'
pip_input : '000'
pbp_input : '000'
pip_size : '001'
pip_position : '000'
msigd: Error querying device on osd_language - got '5b00__003_'
Try 'msigd --help' for more information.
osd_transparency : '000'
osd_timeout : '020'
sound_enable : '000'
rgb_led : '000'
navi_up : '002'
navi_down : '001'
navi_left : '003'
navi_right : '005'
```

Tombana, 2021-03-26:
I got the same monitor, I only tried brightness and contrast settings, those work fine :+1:. Thanks for this great tool @couriersud !
I noticed that the `msigd` query output shows `Monitor Series: MAG321CQR` but it should be `MPG341CQR`.

couriersud, 2021-05-01:
This monitor uses a separate USB device to control LEDs. The work and options supported are documented here:

https://github.com/couriersud/msigd/issues/16
With the next commit I will add support for this monitor. Can you please rerun `msigd --info --debug --query`?

Tombana, 2021-05-03:
Thanks for the update @couriersud .

This is the output of `msigd --info --debug --query` on commit 74090bd:
[query_result.txt](https://github.com/couriersud/msigd/files/6416424/query_result.txt)

It correctly detects `Monitor Series: MPG341 Series` :+1:

couriersud, 2021-05-04:
Thanks a lot! There are quite a number of errors in the result file. I need to compare settings 1:1 to manual. Marked as bug.

Tombana, 2021-05-05:
Let me know if there are any more commands to run. For me the tool already does all that I need.

(Unfortunately I don't have time to dive into the 'usb capture' things that you explained in #16..)

couriersud, 2021-05-08:
Can you please test again using the "multi-monitor" branch. Thanks a lot!

couriersud, 2021-05-08:
Thanks a lot. Which osd language are you using? Can you please set the language to anything different to English and post the results again?
Can you also please confirm that for the PBP modes (X2,X3,X4) only three sources are available?

couriersud, 2021-05-08:
I have committed fixes as well to fix game_mode and osd_language now.

Tombana, 2021-05-08:
OSD on English:
`msigd: Error querying device on osd_language - got '5b00__001_'`
OSD on Dutch:
`msigd: Error querying device on osd_language - got '5b00__010_'`

> Can you also please confirm that for the PBP modes (X2,X3,X4) only three sources are available?

I'm not 100% sure what I should look at. In the monitor menu, when I navigate to "PBP (X2)" I can select "Source 1, Source 2" (which I can then set to HDMI, DP, USB-C, etc). When I go to "PBP (X3)" there are 3 sources and at "PBP (X4)" there are 4 sources.

couriersud, 2021-05-08:
OSD on Dutch? Dutch is not even listed as a supported language even in the Dutch manual:

https://download.msi.com/archive/mnu_exe/monitor/MPG341CQRv1.0_Dutch.pdf

On which position is Dutch in the OSD language menu? English is on 1 (Chinese is 0) and my guess from your input Dutch should be on position 10.

I have also reenabled pbp_input4 in the last commit. Would be great if you could post `msigd -q --info --debug` again.

Tombana, 2021-05-08:
It looks like `pbp_input4` doesn't work, this is on the last commit: [query_pbp_input4.txt](https://github.com/couriersud/msigd/files/6446518/query_pbp_input4.txt)

Dutch is on position 16. Was that `10` hexadecimal maybe?

couriersud, 2021-05-08:
Thanks for confirming that `pbp_input4` does not work. Will remove this again.
When I select position 10 (Simplified Chinese) I get 
`osd_language : '00:'`
Here the language is encoded as ORD('0')+position
Looks like on the MPG341 the encoding is hex. In addition, the reply format is not consistent with any other setting. I consider this as a bug. The `_` (underscore) represent bytes with value 0 and the value starts one position earlier than for other settings.
I also noticed that the response for `screen_assistance` and alarm_clock_time seem to be broken:
`screen_assistance : '__0'`
`alarm_clock_time : '_00'`

can you please try the following:

`./msigd --alarm_clock 1 --alarm_clock_time 20`
Should show an alarm clock

`./msigd --osd_language 16`
Should switch OSD language to Dutch.

`./msigd --screen_assistance red1`
Should show a red cross on screen

Tombana, 2021-05-08:
`./msigd --alarm_clock 1 --alarm_clock_time 20`

This does show a clock counting down, but it starts from `50:00` or something, not 20. Also when I run the command again but with another time, it starts counting down from the same time.

`./msigd --osd_language 16`

This is an interesting one: it completely crashed my monitor. The alarm clock started flickering. The 'joypad' doesn't work anymore. Even the on-off button does not work anymore.
If I try to run anything else, `msigd` says `Error on device identification.`.

couriersud, 2021-05-08:
Unplug your monitor.

Tombana, 2021-05-08:
I took out all the cables, but I think its bricked. When I only plug in the power cable and nothing else, then it shows the MSI logo for a second and then just a black screen. The joypad does not work and the menu button also does not work. I can't find any sort of "factory reset" button on it.
I think this should be covered by warranty. If I get a new one I'm afraid that I won't try out more settings through your tool though. (Its not the fault of your great tool, its really crappy firmware by MSI that can't handle an invalid language value, )

Thanks for all the effort :)

couriersud, 2021-05-08:
I really feel sorry for what happened to you. I ran into similar situations, unplugging helped.Please keep us updated here on the progress with the warranty. I will remove support for osd_language as well as screen_assistance and alarm_clock_time. I will also disable setting values for unknown monitors.
One last try might be to try to reach the service menu ...

```
### 2.2. Service menu

The panel information and more is displayed by the service menu. 

-	Hold the joystick button down
-	Unplug your monitor power supply - keep holding down
-	Plug in your monitor supply - keep holding down
-   When msi logo appears - stop holding down
-	When monitor displays screen - push joystick button down to show OSD
-   Push joystick button down again
-	Service menu opens

The service menu also has more information about the preset color
temperature modes. 
```

Tombana, 2021-05-08:
Thanks for the support. The "holding down while unplugging and plugging in" did not seem to help. I noticed that when I plug it in without holding the joystick, then I get the MSI logo and then it goes blank. If I plug it in while holding the joystick then it does *not* show the MSI logo and directly goes blank. In both cases its not possible to get into any kind of OSD menu. I've also tried the MSI tool from Windows, which also can't detect the monitor.

couriersud, 2021-05-08:
What a nightmare ... Lesson learned, no more new monitors without full usb logs containing the communication between the MSI Gaming app and the monitor. 
I am deeply frustrated. This should not have happened.

Tombana, 2021-05-08:
> I am deeply frustrated. This should not have happened.

Don't worry about it :) I don't blame you in any way. The way I see it, this is a bug in MSI's firmware: something as simple as an invalid language value should not brick the monitor. The monitor should just have rejected the value.

> Lesson learned, no more new monitors without full usb logs containing the communication between the MSI Gaming app and the monitor.

Good idea, then maybe my broken monitor might save other people's monitors from the same fate.

I'll start the warranty process tomorrow and I'll keep you updated on the progress. I'm optimistic that I should be able to get it replaced.

Tombana, 2021-05-14:
Great news: the bricked monitor is sent back for reparation and they've sent me a brand new one which just arrived, free of charge.

closed 2022-11-27 by couriersud