couriersud/msigd #23: DEBUG: Initializing HID lib msigd: No usb device found - Prestige PS341WU

Zackhardtoname, 2020-12-10:
> ➜  msigd git:(master) ✗ ./msigd                       
> msigd: No usb device found
> Try 'msigd --help' for more information.
> ➜  msigd git:(master) ✗ ./msigd           
> ➜  ./msigd --debug --info --query
> ➜  msigd git:(master) ✗ ./msigd --debug --info --query
> DEBUG: Initializing HID lib
> msigd: No usb device found
> Try 'msigd --help' for more information.
> ➜  msigd git:(master) ✗ xrandr                             
> Screen 0: minimum 8 x 8, current 8960 x 2160, maximum 32767 x 32767
> DP-0 disconnected (normal left inverted right x axis y axis)
> DP-1 disconnected (normal left inverted right x axis y axis)
> HDMI-0 disconnected (normal left inverted right x axis y axis)
> DP-2 connected primary 5120x2160+0+0 (normal left inverted right x axis y axis) 800mm x 330mm
>    3840x2160     60.00 +  59.94    50.00    29.97    25.00    23.98    23.98  
>    5120x2160     60.00*   59.94    50.00    50.00    30.00    29.97  
>    3440x1440     59.97    49.99    30.00  
>    1920x1080     60.00    59.94    50.00    29.97    25.00    23.98  
>    1680x1050     59.95  
>    1600x1000     59.87  
>    1280x1024     60.02  
>    1280x960      60.00  
>    1280x800      59.81  
>    1280x720      60.00    59.94    50.00  
>    1152x864      59.96  
>    1024x768      60.00  
>    800x600       60.32  
>    720x480       59.94  
>    640x480       59.94    59.93  
> DP-3 disconnected (normal left inverted right x axis y axis)
> DP-4 disconnected (normal left inverted right x axis y axis)
> DP-5 disconnected (normal left inverted right x axis y axis)
> USB-C-0 connected 3840x2160+5120+0 (normal left inverted right x axis y axis) 597mm x 336mm
>    3840x2160     60.00*+  29.98  
>    2560x1440     59.95  
>    1920x1200     59.88  
>    1920x1080     60.00    59.94    50.00    23.98  
>    1680x1050     59.95  
>    1600x1200     60.00  
>    1280x1024     75.02    60.02  
>    1280x800      59.81  
>    1280x720      60.00    59.94    50.00  
>    1152x864      75.00  
>    1024x768      75.03    60.00  
>    800x600       75.00    60.32  
>    720x576       50.00  
>    720x480       59.94  
>    640x480       75.00    59.93    59.94  

Well it used to work on my Manjaro install. Recently I've switched to arch on two machines and it doesn't work on either of them... I used to only get the problem when I forgot to plug in the USB cable. I have tried many different ports. It also works in Windows.

couriersud, 2021-05-01:
This may be security issue. Please try `sudo ./msigd --debug --info --query`. If this works, you should follow section 4.1 (Linux-security) in README.md.

andbez, 2022-03-02:
Same here on MacOS (12.2.1). Output of sudo ./msigd --debug --info --query

DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A000000000>
msigd: Error on device identification
Try 'msigd --help' for more information.

couriersud, 2022-11-26:
The error means that the tool can not read parameters 00140 and 00150 from the monitor. This is weird. According to the msi website there is firmware `V030` available. Does this happen withe the V030 firmware as well?