couriersud/msigd #24: Support for MAG274QRF-QD

RodoMa92, 2021-03-07:
This is the debug output coming from this monitor, MSI MAG274QRF-QD, firmware FW.011, panel model AUO_M270DAN08_2.

Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         ---
Monitor Series: MAG321CQR
s140:           <00e>
s150:           <V43>
DEBUG: Special 01 b0: 01 5a 40 05 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 12 d3 23 75 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '010'
serial : '---'
frequency : '165'
game_mode : '000'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
zero_latency : '001'
screen_size : '000'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '000'
contrast : '070'
sharpness : '001'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '001'
input : '002'
pip : '000'
pip_input : '002'
pbp_input : '002'
pip_size : '002'
pip_position : '000'
osd_language : '004'
osd_transparency : '000'
osd_timeout : '010'
sound_enable : '001'
rgb_led : '000'
navi_up : '001'
navi_down : '006'
navi_left : '007'
navi_right : '005'

couriersud, 2021-05-02:
Added support for the monitor. Can you please post the debug output again?

RodoMa92, 2021-05-05:
Sorry for the delay, and thanks for the update; here is the output from the command

`DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <--->
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         ---
Monitor Series: MAG274 Series
s140:           <00e>
s150:           <V43>
DEBUG: Special 01 b0: 01 5a 40 05 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 12 d3 23 75 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '001'
mode : '010'
serial : '---'
frequency : '165'
game_mode : '000'
unknown210 : '009'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '000'
zero_latency : '001'
screen_size : '000'
night_vision : '000'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '000'
contrast : '070'
sharpness : '000'
color_preset : '001'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '001'
input : '002'
osd_language : '004'
osd_transparency : '000'
osd_timeout : '010'
sound_enable : '001'
rgb_led : '000'
navi_up : '001'
navi_down : '006'
navi_left : '007'
navi_right : '005'
`

I already added support for this panel, but I've completely forgot to send it back. The section with the PIP support was extremely interesting, but it doesent work. Tried to enable on the panel, but the OSD completely freaks out and will not show up until a complete cold reset of it (it was literally appearing for a quarter of a second before disappearing again); probably code left there in the protocol but disabled in the monitor firmware itself. Such a shame.

Thanks a ton for it, I was interested mainly for the custom button on the left, since for basically all other commands DDC is still a option for this panel.

RodoMa92, 2021-08-17:
A new firmware FW.015, released a few days ago but internally dated 10/05/2021, has been released for this panel, finally adding sRGB, Display P3 and Adobe RGB in the PRO section of this monitor, properly clamping the huge color gamut that this panel have. Apparently now the protocol changed again, since it's unrecognized.

Output of the debug command:

DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <xxx>
Unexpected id combination <00e><V48>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         xxx
Monitor Series: Unknown Series
LED support:    None
s140:           <00e>
s150:           <V48>
DEBUG: Special 01 b0: 01 5a 40 05 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 12 d3 23 75 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
serial : 'xxx'
frequency : '165'
response_time : '001'
image_enhancement : '000'
brightness : '015'
contrast : '070'
sharpness : '000'
color_rgb : 'bbb'
osd_transparency : '000'
osd_timeout : '020'

RodoMa92, 2021-08-21:
I've just completely added support for both firmwares, in theory, as of pull request #31; the only issue now is an issue with the screen assistance command, the comment over the setting explains it better;

As far as I can see almost everything should be there, but as soon as I can I'll play around with the software under a VM to see if they added anything more or if I can understand why screen assistance has that problem.

muhdiboy, 2021-11-18:
Although I've got the MAG274QRFDE-QD model, which is the german version of the same monitor, it is not being recognized.

Are you aware of this and do you plan to support this model too?
If I can help to participate in any form, feel free to contact me.

Here is some info.

```
muhdi@pop-os:~/git/msigd$ lsusb
Bus 004 Device 003: ID 2109:0815 VIA Labs, Inc. USB3.0 Hub
Bus 004 Device 002: ID 2109:0815 VIA Labs, Inc. USB3.0 Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 010: ID 046d:c53f Logitech, Inc. USB Receiver
Bus 003 Device 009: ID 062a:4101 MosArt Semiconductor Corp. Wireless Keyboard/Mouse
Bus 003 Device 008: ID 04f3:0c28 Elan Microelectronics Corp. ELAN:Fingerprint
Bus 003 Device 007: ID 2109:2815 VIA Labs, Inc. USB2.0 Hub
Bus 003 Device 006: ID 1bcf:0215 Sunplus Innovation Technology Inc. Aukey-PC-LM1E Camera
Bus 003 Device 005: ID 05ac:0250 Apple, Inc. Aluminium Keyboard (ISO)
Bus 003 Device 003: ID 2109:2815 VIA Labs, Inc. USB2.0 Hub
Bus 003 Device 012: ID 1462:3fa4 Micro Star International MSI Gaming Controller
Bus 003 Device 011: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 594d:4409 YMDK YMD09
Bus 001 Device 003: ID 1462:7c56 Micro Star International MYSTIC LIGHT
Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 1235:8211 Focusrite-Novation Scarlett Solo (3rd Gen.)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I do also have a MSI motherboard, which could be one if not both of those entries with MSI/Micro Star above.
```
muhdi@pop-os:~/git/msigd$ ./msigd -l
1,,,,/dev/hidraw1
2,,,,/dev/hidraw1
msigd: No usb device found
Try 'msigd --help' for more information.
```

RodoMa92, 2021-11-26:
Yes, please post the output from
`msigd --debug --info --query`
I need the hex values returned from 
s140:   
s150:   
From the output.

muhdiboy, 2021-12-01:
Not much different result :confused:

```
muhdi@pop-os:~/git/msigd$ ./msigd --debug --info --query
DEBUG: Initializing HID lib
msigd: No usb device found
Try 'msigd --help' for more information.
```

RodoMa92, 2021-12-23:
Whoops, my fault. 

Udev needs an additional .rules file in /etc/udev/rules.d in the folder inside udev, the file named 51-msi-gaming-device.rules, to allow access as a normal user, then run as root `udevadm control --reload-rules && udevadm trigger` if you don't want to reboot the machine. Otherwise you need to run this with root access to see it. Then post the last command again. 

Sorry for the late reply.

muhdiboy, 2021-12-23:
Thank you for your support. I guess this is somewhat on my behalf. The README clearly says that I had to do that initially on Linux.

Here is my output now:
```
muhdi@pop-os:~/git/msigd$ ./msigd --debug --info --query
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG274 Series
LED support:    Mystic
s140:           <00e>
s150:           <V43>
DEBUG: Special 01 b0: 01 5a 40 05 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 f1 17 cc 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '000'
serial : 'CA8A271200418'
frequency : '165'
game_mode : '000'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '000'
zero_latency : '000'
screen_size : '000'
night_vision : '001'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '010'
contrast : '070'
sharpness : '001'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
input : '002'
osd_language : '001'
osd_transparency : '001'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '001'
navi_up : '001'
navi_down : '001'
navi_left : '005'
navi_right : '002'
```

IT'S WORKING. I already checked which setting for turning on/off Anti Motion Blur is. It's "--enable_dynamic", which also worked seemlessly.

It would be beneficial to add that info to the README (and the code).
Also I guess you can add this model to the list of functioning monitors. Or does this need further testing of all settings and options?

RodoMa92, 2022-01-02:
Yes, your monitor has the same for Firmware V11 of my panel, so there isn't anything to be added. It should work on almost all the features (besides a small bug for the crosshair selection, haven't yet checked why).

As soon as @couriersud will merge #31, if it's good enough, I guess we can close this issue :eyes:

couriersud, 2022-11-27:
I have merged #31. Closing this

closed 2022-11-27 by couriersud