couriersud/msigd #26: Support for MAG272QR

usrErr0r, 2021-04-07:
I have started to try and get the MAG272QR to work but I have hit a block at the Mystic lights part of the code. The MAG272QR has 12 LEDs as far as GamingOSD's Mystic Lights program can control however I notice from images that a lot of the pre-existing supported keyboards only have 9ish. Also none of the Mystic Light related commands work for me...this is the main problem due to my end goal.

For the record I added another line in the main CPP file saying 
{ MAG272, "00G", "V18", "MAG272 Series", LT_MYSTIC }, // MAG272QR

This makes the program detect the monitor fine and allow me to run commands, a lot of which work but the RGB led related ones do not.

For transparency my goal is to integrate this plugin into OpenRGB somehow and then be able to use it in Artemis RGB or Aurora RGB for effects. I hope somebody can help me fix my problems with getting the commands working, it would be greatly appreciated. I can supply wireshark logs etc. if required and helpful. Thanks! (And great work to the dev so far, program seems very useful)

couriersud, 2021-05-01:
The datasheet at https://storage-asset.msi.com/datasheet/monitor/global/Optix-MAG272QR.pdf does not mention Mystic support. Makes me a bit puzzled. The simplest way would be to start a wireshark log and change the led color from the Gaming OSD program.

couriersud, 2021-05-01:
Forgot: I have added the monitor to msigd. Thanks a lot for your support. You mention that some commands are not working. Would be great to know which.

usrErr0r, 2021-06-30:
Hi there, the limited commands I tested were mainly related to mystic light controls, of which nothing works. I can do wireshark logs, I would just need to double check exactly what you would require

couriersud, 2022-11-26:
I need the size of hid feature reports send to and received from the device. 
I recently bought a MPG273CQR and based on new information rewrote the mystic code. You may  try using `LT_MYSTIC_OPTIX` for your monitor and see if this works.

usrErr0r, 2022-11-26:
@couriersud do you have a link to a build for windows for the latest build? I do not currently have all the pre-requisites for building installed. If not I will try to remember to sort this at a future date. Thanks for getting back to me!

couriersud, 2022-11-27:
No link available. I am using git workflow to test compiling so if you have information on how I could keep the executable files created in the workflow I would implement this.

couriersud, 2022-12-04:
Windows executable can be downloaded here: https://github.com/couriersud/msigd/suites/9658527538/artifacts/462074831

can you please run `msigd --info --debug --query` and `msigd --dump_mystic`?

couriersud, 2023-04-10:
@usrErr0r : Any updates on the mystic support? Without the information I need to disable mystic support for this monitor.
Thanks a lot for your support.