couriersud/msigd #27: Change settings for multiple monitors

PChild, 2021-04-17:
Is there a way to change which monitor is addressed? I have 3 MAG321CURV and only the first screen works with the tool.

The MSI Gaming OSD app has buttons in top left to change which monitor is addressed, I can capture USB traffic if it would help to get this implemented.

[image]

sgtcoder, 2021-04-30:
I agree, I am having this issue too. I have 4 monitors and it only changes one of them. I have to keep all unplugged except the one I want to change, which is a hassle since it moves my monitor alignments.

couriersud, 2021-05-01:
I have added multi-monitor support now. On Windows use `msgid --debug -s unknown` to get a list of connected monitors and their serial numbers. On Linux, use `lsusb -v` to identify the serial number.
Than use `-s SERIAL` or `--serial SERIAL` to specify which monitor to use.

sgtcoder, 2021-05-02:
Thank you. It looks like it's getting there. Now I can plug in all 4 and configure one monitor, but still having issues.

All 4 monitors have the same `iSerial` when using `lsusb -v` unless I am missing something?

couriersud, 2021-05-03:
Can you please recompile to use the USB interface:
```
make clean
make USE_HIDAPI=0
```
Please post the output of the following command here:
```
sudo msgid --debug -s unknown
```

There is another way I can proceed but it needs some more modifications to the source and take some time.

sgtcoder, 2021-05-03:
After running the commands and also this (note misspelling in msgid):

```
make clean
make USE_HIDAPI=0

sudo msigd --debug -s unknown
```

I get this

```
DEBUG: Initializing HID lib
msigd: No usb device found
Try 'msigd --help' for more information.
```

Is there a way we can list the devices and just use the actual serial number? This is what liquidctl does.

PChild, 2021-05-03:
Thanks for the work on this!

I'm also getting identical serial numbers on Windows. I pulled from git, ran `make clean` and `make USE_HIDAPI=0` and then get this when I try to check serial numbers:

```
$ ./msigd --debug -s unknown
DEBUG: Initializing usb_lib
DEBUG: Scanning USB devices...
DEBUG: Found device <MSI Gaming Controller> with serial <A02019010700>

DEBUG: Found device <MSI Gaming Controller> with serial <A02019010700>

DEBUG: Found device <MSI Gaming Controller> with serial <A02019010700>

DEBUG: Endpoint 1 does not exist
msigd: No usb device found
Try 'msigd --help' for more information.
```

couriersud, 2021-05-04:
> After running the commands and also this (note misspelling in msgid):
> 
> ```
> make clean
> make USE_HIDAPI=0
> 
> sudo msigd --debug -s unknown
> ```
> 
> I get this
> 
> ```
> DEBUG: Initializing HID lib
> msigd: No usb device found
> Try 'msigd --help' for more information.
> ```

This is weird. The command is working for me. Just to make sure sudo doesn't pick an executable on the path:

```

> Is there a way we can list the devices and just use the actual serial number? This is what liquidctl does.

couriersud, 2021-05-04:
> Thanks for the work on this!
> 
> I'm also getting identical serial numbers on Windows. I pulled from git, ran `make clean` and `make USE_HIDAPI=0` and then get this when I try to check serial numbers:

Great. Thanks a lot!

sgtcoder, 2021-05-04:
How though? The command you posted is not spelled correctly.

```sudo msgid --debug -s unknown```

It should be `msigd` not `msgid`

I don't need sudo. I have udev rules and it's in `/usr/bin`

sgtcoder, 2021-05-04:
Also 3 references to `msgid` in the readme.md too that should be updated to `msigd`

couriersud, 2021-05-04:
I have create a branch multi-monitor and started work on implementing better support for multi-monitors. Currently, one additional functionality is supported: Listing monitors suitable for msigd.

Can all of you try (after pulling the multi-monitor branch) running

```
./msigd --list
```
and post the output here?

couriersud, 2021-05-04:
> Also 3 references to `msgid` in the readme.md too that should be updated to `msigd`

Fixed in multi-monitor branch. Thanks a lot!

sgtcoder, 2021-05-05:
Here is the output.
```
msigd --list

</dev/hidraw12> <A02019010700> <0> <MS> <MSI Gaming Controller>
</dev/hidraw13> <A02019010700> <0> <MS> <MSI Gaming Controller>
</dev/hidraw14> <A02019010700> <0> <MS> <MSI Gaming Controller>
</dev/hidraw15> <A02019010700> <0> <MS> <MSI Gaming Controller>
```

PChild, 2021-05-06:
Output on Windows:

```
$ ./msigd -l
<019> <A02019010700> <0> <MS> <MSI Gaming Controller>
<017> <A02019010700> <0> <MS> <MSI Gaming Controller>
<002> <A02019010700> <0> <MS> <MSI Gaming Controller>
```

couriersud, 2021-05-06:
Thanks a lot for the feedback. To cut a long story short: MSI got it wrong by being too lazy to assign **unique** serial numbers to monitors. That is what the usb standard assumes, that is what the udev subsystem assumes and sysfs as well:

`/sys/bus/hid/devices/0003\:1462\:3FA4.00EE/input/input299/uniq`

On linux using the usb subsystem (and not the preferred libhid) I could use sysfs - or provide instructions on how to use shell scripts to to get the device number to bus and port, e.g.

`grep -R "MSI Gaming Controller" /sys/bus/usb/devices/*/product | cut -f 6 -d /`
`cat /sys/bus/usb/devices/1-7.3/devnum`

This is awkward. No idea as well on how to do this for Mac and Windows or for HID. Please read the security section in Readme.md for more details on why HID is preferred.

The best I can offer for now is use a logic monitor id, e.g. `msigd --lmid 1 -q`. This will depend on the order in which the usb devices either using usb or hid are enumerated. But it is the best what is available cross-platform.

libusb.h may be an alternative - but this is a longer term project since it needs some refactoring of the code first. It offers access to the usb port number which should not change between reboots.

sgtcoder, 2021-05-06:
@couriersud Thank you for the responses. I am sure we can look into it deeper too and see what can be done.

I also use `liquidctl` which is also supported on Windows, Mac, and Linux. They use Python, but I am sure it's comparable to C++, but they are getting the actual serial numbers from the monitor (just like you show it under the `--query`.

Maybe some ideas can come from there: https://github.com/liquidctl/liquidctl

couriersud, 2021-05-06:
Can you please provide the liquidctl configuration for your msi monitor and the output it gives for your msi monitor?
In addition, please provide the output of `msigd -q`.

sgtcoder, 2021-05-06:
Sorry, liquidctl, it's not for the monitor. I use it for my AIO and 2 Smart Hubs. All 3 of them are USB, but instead of using iSerial they are using the actual serial numbers, if I am understanding correctly.

```
msigd --query

macro_key : off
mode : user
serial : CA8A220C00608
frequency : 165
game_mode : user
response_time : fast
enable_dynamic : on
hdcr : off
refresh_display : off
refresh_position : left_top
alarm_clock : off
alarm_clock_index : 0
alarm_clock_time : 
alarm_position : left_top
screen_assistance : off
free_sync : off
zero_latency : on
screen_size : 19
pro_mode : user
eye_saver : off
image_enhancement : off
brightness : 60
contrast : 70
sharpness : 0
color_preset : custom
color_red : 100
color_green : 94
color_blue : 96
color_rgb : 
unknown435 : 001
input : dp
pip : off
pip_input : dp
pbp_input : dp
pip_size : large
pip_position : left_top
osd_language : 1
osd_transparency : 0
osd_timeout : 20
sound_enable : on
rgb_led : on
navi_up : game_mode
navi_down : screen_assistance
navi_left : alarm_clock
navi_right : input
```

As you can see the serial number in `msigd --query` is different from `lsusb -v` `serial : CA8A220C00608`. So couldn't we use the serial number of the monitor instead of the iSerial number from `lsusb -v`?

sgtcoder, 2021-05-06:
Actually, after looking at liquidctl, they are also using the iSerial, but NZXT actually puts the serial numbers in iSerial.

Either way, how are you getting the serial in `msigd --query` That might be a way to reference the monitor. I might have some general ideas.

sgtcoder, 2021-05-06:
Does the Bus ID change/unique? `Bus 001 Device 020: ID 1462:3fa4 Micro Star International`

couriersud, 2021-05-06:
The issue is that my msi monitor does not report a serial number. The serial number string contains only blanks. So even this is not consistently implemented.
Apart from that msigd would have to open each device and query the serial number. That would slow down the execution.

sgtcoder, 2021-05-06:
What information do you get without opening up the device to query? I don't think you do a grep (search) on `lsusb -v`?

couriersud, 2021-05-06:
No I don't use grep. Whatever I will do needs to be cross-functional. So any information for identifying the monitor needs to be provided by the interface used without opening the device. MSI just got it wrong by not providing a unique serial number in iSerial.
I will proceed with the enumeration approach and logic monitor numbers from 1 to n.

sgtcoder, 2021-05-06:
That makes sense. I am sure something is comparable. But as you said, you have an approach. We are here if needed. I am working on a move to my own place next month.

couriersud, 2021-05-07:
I have updated the multi-monitor branch. Here's the commit message

```
Add --monitor option to solve issue #27

Use --list to list all available monitors. The first entry in the comma
separated list is the logical monitor number. Use this number as an
argument to --monitor/-m to specify the target monitor.
```

Use `./msigd -m 1 -q` to query the first monitor, `msigd -m 2 -q`  to query the second monitor and so on.

Feedback much welcome, especially from windows and mac users.

sgtcoder, 2021-05-07:
Wow, works great! Wrote a script to loop over the 4 monitors I have to update the settings, and every monitor updated!

couriersud, 2021-05-07:
Can you please check if the monitors are still listed in the same order after a reboot or plug/unplug event?

PChild, 2021-05-08:
Confirming that this is working great on Windows! I'll test reboot / replug tomorrow to see if the ordering is the same.

sgtcoder, 2021-05-08:
Same here, will hopefully check tomorrow (for Linux).

couriersud, 2021-05-08:
Thanks a lot - this is really good news. Now let's hope than Windows behaves in the same way.

sgtcoder, 2021-05-08:
Well shoot, I get better support here than anything. So thank you.

My MSI Board BIOS rarely ever pushes the display upon boot. You would think since it's all the same company it would work! That and my monitors never stay in order and I have to constantly reconfigure the refresh rate and order, whereas my old ASUS ones saved everything. So just been dealing with that.

sgtcoder, 2021-05-08:
I have a two ideas to add (no rush).

1. Now that we have `msigd --list`, isn't it possible to pass the actual serial number now, and you would just query the devices from `msigd --list` to get the USB Bus ID to apply the commands to?

2. I noticed there is no way to disable auto-scan on the monitor via `msigd`. Is that possible to add? I am not sure if auto scan is causing my display issues with the BIOS (but something is going on with the monitor receiving signal or waking up).

couriersud, 2022-11-26:
> 1. Now that we have `msigd --list`, isn't it possible to pass the actual serial number now, and you would just query the devices from `msigd --list` to get the USB Bus ID to apply the commands to?

The `--list` option provides the **USB serial**. This is the same on my MAG321CURV and my MPG273CQR. This seems to be the same for all monitors :-( The serial provided by the usb interface with command code `00130` doesn't work with all monitors. My MAG321CURV returns just blanks. You can use a script though : get the list of monitors, query the serial for each monitors, match to the logical sequence id and use this going forward.

> 2\. I noticed there is no way to disable auto-scan on the monitor via `msigd`. Is that possible to add? I am not sure if auto scan is causing my display issues with the BIOS (but something is going on with the monitor receiving signal or waking up).

If you can't disable it with the monitor OSD there is no way to disable it with `msigd`. My more recent `MPG273CQR` has an option to disable auto scan and for this monitor `msigd` now has a setting.

closed 2023-05-18 by PChild