couriersud/msigd #28: Support for MAG271CR

thorio, 2021-05-06:
<details>
<summary>Query Output</summary>

```
λ msigd --info --debug --query
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <--->
Unexpected id combination <005><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         ---
Monitor Series: MAG321CQR
s140:           <005>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 19 01 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 f0 38 ae 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
mode : '010'
serial : '---'
frequency : '144'
game_mode : '000'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '00l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on screen_size - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on pro_mode - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on eye_saver - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
sharpness : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on color_preset - got ''
Try 'msigd --help' for more information.
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
unknown435 : '000'
input : '002'
msigd: Error querying device on pip - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pbp_input - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_size - got '56006'
Try 'msigd --help' for more information.
msigd: Error querying device on pip_position - got '56006'
Try 'msigd --help' for more information.
osd_language : '001'
osd_transparency : '002'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '000'
navi_up : '001'
navi_down : '002'
navi_left : '003'
navi_right : '005'
```
</details>

Even though the tool reports an error querying brightness and contrast, I can query and set them individually (with `-f` and `--brightness`).
Additionally, whenever I try to query `zero_latency`, I get this:
<details>
<summary>Output</summary>

```
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <--->
Unexpected id combination <005><V18>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
DEBUG: Error 0 reading from HID device
msigd: Error querying device on zero_latency - got ''
Try 'msigd --help' for more information.
```
</details>  

The tool then stops working entirely, with this error:
<details>
<summary>Output</summary>

```
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <--->
DEBUG: Error 0 reading from HID device
msigd: Error on device identification
Try 'msigd --help' for more information.
```
</details>

This is fixable by unplugging and replugging the monitor.

couriersud, 2021-05-07:
Thanks a lot, hope to take care of it over the weekend. Need to finish multi-monitor support first.

thorio, 2021-05-22:
Finally got around to setting up the dev environment again, here are the results:
<details>
<summary>Output</summary>

```
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <--->
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         ---
Monitor Series: MAG271CR Series
LED support:    Mystic
s140:           <005>
s150:           <V18>
DEBUG: Special 01 b0: 01 5a 19 01 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 f0 38 ae 00 00 00 00 00 00 00 00 00 00
macro_key : '000'
serial : '---'
frequency : '144'
game_mode : '000'
black_tuner : '00:'
response_time : '000'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_index : '000'
alarm_clock_time : '0<l'
alarm_position : '000'
screen_assistance : '000'
free_sync : '001'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '000'
brightness : '050'
contrast : '050'
sharpness : '000'
color_preset : '003'
color_red : '100'
color_green : '100'
color_blue : '100'
color_rgb : '0x949494'
input : '002'
osd_language : '001'
osd_transparency : '002'
osd_timeout : '020'
sound_enable : '001'
rgb_led : '000'
navi_up : '001'
navi_down : '002'
navi_left : '003'
navi_right : '005'
```
</details>

Looks to me like everything is working now, thanks!

closed 2021-05-22 by thorio