couriersud/msigd #32: Unknown monitor MAG321CURV

firegore2, 2021-10-20:
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
DEBUG: Special 01 b4: 01 5a 32 d6 05 b2 00 00 00 00 00 00 00 00 00 00
Unexpected id combination <00;><V43>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: Unknown Series
LED support:    None
s140:           <00;>
s150:           <V43>
DEBUG: Special 01 b0: 01 5a 35 03 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 32 d6 05 b2 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
serial : 'DA2A011100337'
frequency : '060'
response_time : '000'
image_enhancement : '001'
brightness : '052'
contrast : '074'
sharpness : '000'
color_rgb : '0x949494'
osd_transparency : '000'
osd_timeout : '020'

Windows 10

couriersud, 2022-11-27:
This is an interesting case. I own the same monitor. This one seems to have a newer firmware ("V43")  than mine ("V18"). It also returns a serial whereas mine returns just blanks. And the support page (https://www.msi.com/Monitor/Optix-MAG321CURV/support#manual) does not provide an english manual. This was available when I bought the monitor some years ago.
Can you provide a screenshot of the OSD Window just after pressing the OSD button, please?

couriersud, 2022-11-27:
Please try branch '32-unknown-monitor-mag321curv' - I have added the monitor there.
When did you buy the monitor? The manual on the msi website is from 2019 and does not mention that monitors with differing specifications exist.

couriersud, 2023-04-10:
Monitor added - please open a new issue in case you encounter problems

closed 2023-04-10 by couriersud