couriersud/msigd #33: MAG321QR Support

sl33nyc, 2021-12-25:
```
└─[0] <git:(master 4b5c4f0✈) > sudo ./msigd --info --debug --query
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Unexpected id combination <00{><V51>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: Unknown Series
LED support:    None
s140:           <00{>
s150:           <V51>
DEBUG: Special 01 b0: 01 5a 46 18 15 00 00 d9 26 00 00 06 58 01 00 00
DEBUG: Special 01 b4: 01 5a 02 90 07 40 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
DEBUG: Error -1 writing to HID device
msigd: Error querying device on macro_key - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on serial - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on frequency - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on response_time - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on image_enhancement - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on contrast - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on sharpness - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on color_rgb - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on osd_transparency - got ''
Try 'msigd --help' for more information.
DEBUG: Error -1 writing to HID device
msigd: Error querying device on osd_timeout - got ''
Try 'msigd --help' for more information.
```

couriersud, 2022-11-27:
Same issue here as for #32 and others. MSI seems to have upgraded software. Contrary to the other issues this seems to be a new hardware release (QR) since "00{" is different to the id "00:" returned by the MAG321CQR.

I will need some time to review the manual and select the right options to add this monitor.

couriersud, 2022-12-01:
The manual is close to the MPG273 I own. It has MPRT, KVM, but not the ambient light adjustments.

I have added experimental support (no setting of parameters) to branch https://github.com/couriersud/msigd/tree/33-mag321qr-support
Can you run `./msigd --info --debug --query` on this branch, please?

Does the monitor has mystic light leds? From the product pictures it looks like it has two groups: The logo with some leds and an array? Can you confirm this?

sl33nyc, 2022-12-02:
> Does the monitor has mystic light leds? From the product pictures it looks like it has two groups: The logo with some leds and an array? Can you confirm this?

I think it only has the array of LEDs but none on the logo.

```
└─[0] <git:(33-mag321qr-support c35449a✈) > ./msigd --info --debug --query 
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: MAG321QR
LED support:    MysticOptix
s140:           <00{>
s150:           <V51>
DEBUG: Special 01 b0: 01 5a 46 18 15 00 00 d9 26 00 00 06 58 01 00 00
DEBUG: Special 01 b4: 01 5a 02 90 07 40 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '010'
serial : 'DB9A031700262'
frequency : '060'
game_mode : '000'
response_time : '002'
hdcr : '001'
refresh_display : '000'
refresh_position : '000'
alarm_clock : '000'
alarm_clock_time : '02l'
alarm_position : '000'
smart_crosshair_icon : '000'
smart_crosshair_color : '002'
free_sync : '001'
screen_size : '000'
night_vision : '000'
pro_mode : '000'
eye_saver : '000'
image_enhancement : '001'
brightness : '070'
contrast : '070'
sharpness : '000'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
input : '000'
auto_scan : '001'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
rgb_led : '001'
power_button : '000'
hdmi_cec : '001'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on ambient_brightness - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on ambient_rgb - got ''
Try 'msigd --help' for more information.
DEBUG: Error 0 reading from HID device
msigd: Error querying device on ambient_brightness_custom - got ''
Try 'msigd --help' for more information.
kvm : '000'
DEBUG: Error 0 reading from HID device
msigd: Error querying device on sound_tune - got ''
Try 'msigd --help' for more information.
navi_up : '002'
navi_down : '003'
navi_left : '009'
navi_right : '005'
```

sl33nyc, 2022-12-02:
```
└─[0] <git:(33-mag321qr-support 2583bac✈) > ./msigd --dump_mystic
Mystic record size: 168

0000: 72 19 ff 00 00 00 64 00 
0008: ff 00 80 00 21 ff 00 00 
0010: 00 64 00 ff 00 80 00 ff 
0018: 00 00 ff 00 00 ff 00 00 
0020: ff 00 00 ff 00 00 ff 00 
0028: 00 ff 00 00 ff 00 00 ff 
0030: 00 00 00 00 00 00 00 00 
0038: 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 
0048: 00 00 00 00 00 00 00 00 
0050: 00 00 00 00 00 00 00 00 
0058: 00 00 00 00 00 00 00 ff 
0060: 00 00 ff 00 00 ff 00 00 
0068: ff 00 00 ff 00 00 ff 00 
0070: 00 ff 00 00 ff 00 00 ff 
0078: 00 00 ff 00 00 ff 00 00 
0080: ff 00 00 00 00 00 00 00 
0088: 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 00 00 00 00 
0098: 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00
```

closed 2022-12-02 by couriersud