couriersud/msigd #38: Unknown monitor Optix MAG241C

Ovenoboyo, 2022-06-05:
I'm not sure why this is happening. Optix MAG241C should show ```<002><V18>``` as ID combination but mine shows as ```<002><V49>```

This monitor was bought in second half of 2022. Maybe its a new revision of MAG241C

Heres the output of ```msigd --info --debug --query```
```msigd --info --debug --query
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02018011700>
Unexpected id combination <002><V49>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02018011700
Monitor Series: Unknown Series
LED support:    None
s140:           <002>
s150:           <V49>
DEBUG: Special 01 b0: 01 5a 15 00 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 02 92 0d 34 00 00 00 00 00 00 00 00 00 00
DEBUG: Error 0 reading from HID device
DEBUG: Error receiving 01 d0 62 30 30 31 30 30 30 30 30 0d: 0
macro_key : '000'
serial : 'EA2T111600179'
frequency : '144'
response_time : '002'
image_enhancement : '000'
brightness : '100'
contrast : '070'
sharpness : '000'
color_rgb : 'bbb'
osd_transparency : '000'
osd_timeout : '020'```

couriersud, 2022-11-27:
Looks like this is connect to issue #32. The website does not provide any indications that something has changed since it was added Could you provide a screenshot of the OSD window after pressing the OSD button, please?

couriersud, 2022-12-03:
Please get branch https://github.com/couriersud/msigd/tree/38-unknown-monitor-optix-mag241c and provide `msigd --info --debug --query` again. Thanks!

Ovenoboyo, 2022-12-12:
Really sorry but I do not have access to my monitor right now. I'll try to update in a few days.

couriersud, 2022-12-15:
First, sorry for the late reply ... I was travelling the last days. 
Within the issue branches I am disabling any setting changes - to reduce the risk of producing bricks to a minimum.

This looks good. I'll review the manual and post an update soon.

couriersud, 2022-12-15:
MAG241C V49 is now supported in the master branch. Would be great if you could provide feedback so that I can close this issue

Ovenoboyo, 2022-12-17:
> First, sorry for the late reply ... I was travelling the last days. 
> Within the issue branches I am disabling any setting changes - to reduce the risk of producing bricks to a minimum.
> 
> This looks good. I'll review the manual and post an update soon.

Oh that's understandable. Thank you for the great work