couriersud/msigd #39: MAG274QRF-QD issue setting crosshair to values >=10

couriersud, 2022-11-27:
Leaving a todo to revisit crosshair setting on MAG274QRF There seems to be an issue with the crosshair setting.
Test crosshair again on my MAG321CURV

couriersud, 2022-11-27:
See also issue #24

couriersud, 2022-12-02:
This has been fixed now in master. Would be great to get feedback for MAG274QRF-QD --> #24

RodoMa92, 2022-12-20:
Just saw this, sorry! I'll check as soon as I can. Thanks for your code :)

couriersud, 2023-01-11:
It would be great if you could post the output of

`./msigd --dump_mystic`

and

`msigd --debug --info --query`

here. The first one is needed to correctly identify the mystic hardware. Thanks a lot!

RodoMa92, 2023-01-12:
`./msigd --dump_mystic`:
`Mystic record size: 168

0000: 72 21 ff 00 00 00 64 00 
0008: ff 00 80 00 21 ff 00 00 
0010: 00 64 00 ff 00 80 00 ff 
0018: 00 00 ff 00 00 ff 00 00 
0020: ff 00 00 ff 00 00 ff 00 
0028: 00 ff 00 00 ff 00 00 ff 
0030: 00 00 ff 00 00 ff 00 00 
0038: ff 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 
0048: 00 00 00 00 00 00 00 00 
0050: 00 00 00 00 00 00 00 00 
0058: 00 00 00 00 00 00 00 ff 
0060: 00 00 ff 00 00 ff 00 00 
0068: ff 00 00 ff 00 00 ff 00 
0070: 00 ff 00 00 ff 00 00 ff 
0078: 00 00 ff 00 00 ff 00 00 
0080: ff 00 00 00 00 00 00 00 
0088: 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 00 00 00 00 
0098: 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00 
`
`DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <xxx>
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         xxx
Monitor Series: MAG274QRF-QD FW.020
LED support:    Mystic
s140:           <00e>
s150:           <V56>
DEBUG: Special 01 b0: 01 5a 40 05 00 00 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 b4: 01 5a 12 d3 23 75 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
mode : '014'
serial : 'xxx'
frequency : '060'
game_mode : '000'
response_time : '001'
enable_dynamic : '000'
hdcr : '000'
refresh_display : '000'
refresh_position : '004'
alarm_clock : '000'
alarm_position : '004'
screen_assistance : '000'
free_sync : '001'
night_vision : '000'
pro_mode : '004'
eye_saver : '000'
image_enhancement : '000'
brightness : '015'
contrast : '070'
sharpness : '001'
color_preset : '001'
color_red : '050'
color_green : '050'
color_blue : '050'
color_rgb : 'bbb'
input : '002'
auto_scan : '000'
osd_language : '004'
osd_transparency : '000'
osd_timeout : '020'
screen_info : '000'
sound_enable : '001'
rgb_led : '000'
power_button : '000'
hdmi_cec : '000'
navi_up : '001'
navi_down : '007'
navi_left : '006'
navi_right : '005'`

Here you go, hope that this contains all the things you need :P

Marco.

RodoMa92, 2023-01-12:
Oh, I actually have tested the previous issue coming from the screen assistance not working. Yes, now it's working 100%, I should have guessed that the problem was the endianness of the data, since it was cutting out exactly at 9, but hey ho, better late than never :P

couriersud, 2023-01-12:
Thanks a lot, updated mystic (168 bytes - optix hardware), closing this nowf