couriersud/msigd #43: MAG274QRX Support

kunver400, 2023-02-24:
MAG274QRX

```
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Unexpected id combination <00|><V53>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: Unknown Series
LED support:    None
s140:           <00|>
s150:           <V53>
DEBUG: Special 01 b0: 01 5a 53 05 15 00 00 d9 26 00 00 b4 f5 01 01 00
DEBUG: Special 01 b4: 01 5a 02 8d 1f 47 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
serial : 'CA8A722300174'
frequency : '240'
response_time : '001'
eye_saver : '000'
image_enhancement : '000'
brightness : '040'
contrast : '070'
sharpness : '000'
color_rgb : '0x949494'
osd_transparency : '000'
osd_timeout : '020'

```

lsusb

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 1462:3fa4 Micro Star International MSI Gaming Controller
Bus 001 Device 009: ID 04d9:0348 Holtek Semiconductor, Inc. Keyboard
Bus 001 Device 007: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 004: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 005: ID 046d:c53f Logitech, Inc. USB Receiver
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 002: ID 0414:a00b Giga-Byte Technology Co., Ltd USB Audio
Bus 001 Device 010: ID 8087:0033 Intel Corp. 
Bus 001 Device 011: ID 048d:5702 Integrated Technology Express, Inc. RGB LED Controller
Bus 001 Device 008: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

OS: Arch linux 6.1.12-arch1-1

couriersud, 2023-04-09:
I started work on this in a new branch: https://github.com/couriersud/msigd/tree/43-mag274qrx-support
I hope to find some time late April to add support.

couriersud, 2023-04-10:
It would be great if you could post the output of `./msigd --dump_mystic` as well.

couriersud, 2023-04-10:
I have added support for the monitor to the branch. 
Could you provide the output of `'msigd --info --debug --query'`, please?

couriersud, 2023-04-10:
I have disabled setting monitor parameters. Once query works, I will enable those.