couriersud/msigd #44: MD272QP support

wpisarski, 2023-02-28:
Listings for MD272QP on Fedora 36, kernel 6.1.10. Monitor plugged in to the PC through USB-B and HDMI. If needed I can provide additional info on USB-C side from a Win10 laptop. Monitor supported by MSI Productivity Intelligence app on Windows.

```
sudo ./msigd --debug --info --query
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Unexpected id combination <00�><V51>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: Unknown Series
LED support:    None
s140:           <00�>
s150:           <V51>
DEBUG: Special 01 b0: 01 5a 52 00 15 00 00 d9 26 00 00 78 ac 01 01 00
DEBUG: Special 01 b4: 01 5a 02 8d 1f a4 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
serial : 'PB1H102800267'
frequency : '075'
response_time : '001'
eye_saver : '000'
image_enhancement : '000'
brightness : '070'
contrast : '070'
sharpness : '000'
color_rgb : 'bbb'
osd_transparency : '002'
osd_timeout : '020
```

Listing for the monitor with mouse and keyboard plugged into the built in hub.
```
sudo lsusb -vd 1462:3fa4

Bus 005 Device 027: ID 1462:3fa4 Micro Star International MSI Gaming Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1462 Micro Star International
  idProduct          0x3fa4 
  bcdDevice            0.00
  iManufacturer           1 MS
  iProduct                2 MSI Gaming Controller
  iSerial                 3 A02019010700
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      95
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

couriersud, 2023-04-09:
Can you please run `./msigd --debug --info --query > out.txt` and upload `out.txt` here? I need the file to get the exact character code for `s140`.
Thanks a lot!

wpisarski, 2023-04-09:
Here: [out.txt](https://github.com/couriersud/msigd/files/11186477/out.txt)

couriersud, 2023-04-09:
Ok, great, the string representation is `"\x30\x30\x85"`. Hex `85` is not a valid UTF8 character :-(
Need to change the code to use a hex dump.

couriersud, 2023-04-10:
I added support for the monitor in this branch:
https://github.com/couriersud/msigd/tree/43-mag274qrx-support

You will not be able to modify monitor settings. We first need to ensure that `./msigd --debug --info --query` works.

wpisarski, 2023-04-15:
Sorry for the late reply, I was away from my monitor. I'm attaching output from that branch:
[out_branch.txt](https://github.com/couriersud/msigd/files/11240872/out_branch.txt)
There are also errors on stderr:
```msigd: Error querying device on navi_up - got '5b00960004'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_down - got '5b00970008'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_left - got '5b00960004'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_right - got '5b00970008'
Try 'msigd --help' for more information.```

couriersud, 2023-04-15:
Thanks a lot!
on stderr I am missing a line like
`msigd: Error querying device on `**`navi_up`**` ...`
Can you please confirm you get this line?

wpisarski, 2023-04-15:
It's there, I forgot that triple backtick reads the first line as language tag here on GitHub and eats it.

```
msigd: Error querying device on navi_up - got '5b00960004'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_down - got '5b00970008'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_left - got '5b00960004'
Try 'msigd --help' for more information.
msigd: Error querying device on navi_right - got '5b00970008'
Try 'msigd --help' for more information.
```

couriersud, 2023-04-17:
I finally managed to get a WIN10 VM up and running. That caused the delay coming back to this issue.
Are you able to control your monitor settings with one of the MSI apps:

- Gaming Intelligence
- An MSI center module
- DisplayKit

Gaming Intelligence does not seem to suport the MD272QP. DisplayKit seems to support most settings based on a string analysis, but I am not able to locate the codes (e.g. 5800500 for input source). In addition DisplayKit does not support my monitor. So it's pretty difficult to get indications which codes are used by the currently unsupported settings.

In the mean time I propose I release the code without support for Navi button and audio volume. Would this be acceptable to you?

wpisarski, 2023-05-25:
Sorry for the late reply, I somehow missed this.

This would be acceptable. I only have a heavily locked-down work Win10 laptop, but I can try setting up a Win10 box on my end and check.