couriersud/msigd #45: New ID for MAG272CQR

AlexFullmoon, 2023-04-02:
Updated firmware, probably?
```
DEBUG: Initializing HID lib
DEBUG: Found <MSI Gaming Controller> with serial <A02019010700>
Unexpected id combination <00E><V41>
Detected an unknown monitor. Please report the output of
'msigd --info --debug --query' as an issue and also provide
the ID (MAG...) of your monitor. Thank you!
Vendor Id:      0x1462
Product Id:     0x3fa4
Product:        MSI Gaming Controller
Serial:         A02019010700
Monitor Series: Unknown Series
LED support:    None
s140:           <00E>
s150:           <V41>
DEBUG: Special 01 b0: 01 5a 38 01 15 00 00 d9 26 00 00 8d bf 01 00 00
DEBUG: Special 01 b4: 01 5a 92 c5 07 9e 00 00 00 00 00 00 00 00 00 00
DEBUG: Special 01 d0 62 30 30 31 30 30 30 30 30 0d: 01 d0 36 30 30 2b 0d 30 30 30 30 0d 00 00 00 00
macro_key : '000'
serial : 'CA6T141C00202'
frequency : '165'
response_time : '001'
eye_saver : '000'
image_enhancement : '000'
brightness : '030'
contrast : '070'
sharpness : '000'
color_rgb : '0x949494'
osd_transparency : '001'
osd_timeout : '020'
```

couriersud, 2023-04-09:
I have add a new branch: https://github.com/couriersud/msigd/tree/45-new-id-for-mag272cqr
I have enabled V41. Would we great if you could test and provide feedback.

couriersud, 2023-04-10:
> Note: currently you have duplicate MAG272CQR entries in help output.

Noted. 

The mystic dump helped. Here 's an example for mystic: `-mystic 1:static:0xff0000,0xffffff` sets first led to red, all others to white. `msigd --help` has more detail, your monitor only has one led group (1).

I have updated the code:

`zero_latency`: Not supported according to monitor manual -> removed.
`mode`: made this read-only. Only a limited number of monitors seem to support this setting. Please use `game_mode` and `pro_mode` to set mode.

Are there any other settings which do not work?

AlexFullmoon, 2023-04-11:
Everything works now, thanks.

Now I just have to compile it for Windows...

closed 2023-04-15 by couriersud