couriersud/msigd #51: Compile errors, MSI MPG341QR on Garuda Linux (arch btw)

neomikr0n, 2023-05-12:
Hey! I got an **MSI MPG341QR** for my Garuda Linux (arch based), so I tried to ***make TARGETOS=arch***, but there are a [bunch of errors](https://github.com/neomikr0n/dotfiles/blob/master/garuda-compile-errors) and doesn't compile.

Already tried:
- make, make TARGETOS=osx (lol)

- make clean, then: make USE_HIDAPI=0 &> [moar-errors](https://github.com/neomikr0n/dotfiles/blob/master/garuda-compile-errors02)

-  reinstalled dependencies:
```
╰─λ sudo pacman -S libusb hidapi
warning: libusb-1.0.26-2 is up to date -- reinstalling
warning: hidapi-0.13.1-2 is up to date -- reinstalling
```

- Installing dev/devel libraries, turns out Arch Linux based distros does not split its packages resulting in:
```
yay -S libusb-dev libhidapi-dev libusb-devel hidapi-devel
 -> No AUR package found for libusb-dev
 -> No AUR package found for libhidapi-dev
 -> No AUR package found for libusb-devel
 -> No AUR package found for hidapi-devel
```

- Editing makefile changing "-lhidapi" to "lhidapi-hidraw" or "lhidapi-libusb" and several other creative options without success.

Could you please kindly help? Thank you in advance! :)

--------------
**P.D. More info:**
-------------- 
**neofetch dixit:**
```
OS: Garuda Linux 
Kernel: 6.3.1-zen2-1-zen 
Uptime: 5 hours, 11 mins 
Packages: 2084 (pacman) 
Shell: bash 
DE: Hyprland 
WM: sway 
Terminal: foot 
CPU: 13th Gen Intel i5-13600K (20) @ 5.2GHz 
GPU: AMD ATI Radeon RX 6650 XT / 6700S / 6800S 
Memory: 11.19GiB / 31.12GiB 
```

**[🔍] × sudo lsusb**
```
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 1462:3fa4 Micro Star International MSI Gaming Controller
Bus 001 Device 007: ID 0bda:4c3a Realtek Semiconductor Corp. USB Audio
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 008: ID 1532:0071 Razer USA, Ltd RZ01-0254 Gaming Mouse [DeathAdder Essential White Edition]
Bus 001 Device 006: ID 08bb:2704 Texas Instruments PCM2704 16-bit stereo audio DAC
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

couriersud, 2023-05-22:
Thanks for reporting this. Missing cstdint header which on all other compilers seems to be implicitly being included - but not on your linux flavour.
I have created a separate branch for testing: https://github.com/couriersud/msigd/tree/51-compile-errors-msi-mpg341qr-on-garuda-linux-arch-btw
This should compile now. Please follow the instructions from the Readme.md for arch-type distros.

neomikr0n, 2023-05-23:
I followed your instructions, [this is the log](https://github.com/neomikr0n/dotfiles/blob/master/garuda-compile-errors03). Basically:

1. Compiled with make 'TARGETOS=arch', resulting in 3 warnings.
2. Tried to recompile again, same command, but now is up-to-date, doing nothing.
3. Running 'msigd --help' shows isn't installed yet.

If there's something I'm missing, just let me know and will try it, thanks for your work!

SirDeadlystrike, 2023-10-01:
@couriersud I think this needs to be merged.

I am seeing both local and github workflow build failures when targeting windows from the master branch however your https://github.com/couriersud/msigd/tree/51-compile-errors-msi-mpg341qr-on-garuda-linux-arch-btw branch fixes these errors for windows.

RodoMa92, 2024-03-25:
Same here on arch linux, the header is necessary to make it compile again.