couriersud/msigd #63: Fedora libusb-devel dependency

NostalgiaRealm, 2025-01-09:
Before I used Ubuntu based Pop!_Os and was able to install the dependencies and compile this program normally. Now I use Fedora 41 and tried installing the dependencies before compiling. However, trying so yields the following:
```
rick@MAIN-PC:~$ sudo dnf install libusb-devel hidapi-devel
[sudo] password for rick: 
Updating and loading repositories:
Repositories loaded.
Failed to resolve the transaction:
No match for argument: libusb-devel
Package "hidapi-devel-0.14.0-5.fc41.x86_64" is already installed.
You can try to add to command line:
  --skip-unavailable to skip unavailable packages
```

The libusb-devel package seems to be MIA, trying to compile the program using make yields a bunch of errors. I am guessing it is because of the missing dependency. I hope the readme can be changed to include the correct dependency.
```
rick@MAIN-PC:~/msigd$ make
Compiling src/msigd.cpp...
In file included from src/msigd.cpp:7:
src/phid.h: In static member function ‘static int usbdev_t::get_device_list(logger_t&, unsigned int, unsigned int, device_info_list&)’:
src/phid.h:63:46: warning: unused parameter ‘logger’ [-Wunused-parameter]
   63 |         static int get_device_list(logger_t &logger, unsigned idVendor, unsigned idProduct,
      |                                    ~~~~~~~~~~^~~~~~
src/phid.h: In member function ‘int usbdev_t::control_msg_write(int, int, int, int, void*, int, int)’:
src/phid.h:91:44: warning: unused parameter ‘timeout’ [-Wunused-parameter]
   91 |                 void *bytes, int size, int timeout)
      |                                        ~~~~^~~~~~~
src/phid.h: In member function ‘int usbdev_t::control_msg_read(int, int, int, int, void*, int, int, int&)’:
src/phid.h:111:44: warning: unused parameter ‘timeout’ [-Wunused-parameter]
  111 |                 void *bytes, int size, int timeout, int &ret_size)
      |                                        ~~~~^~~~~~~
In file included from src/msigd.cpp:12:
src/psteelseries.h: At global scope:
src/psteelseries.h:22:32: error: expected ‘)’ before ‘n’
   22 |         steel_rgb_entry(uint8_t n, uint8_t vr, uint8_t vg, uint8_t vb)
      |                        ~       ^~
      |                                )
src/psteelseries.h:29:22: error: ‘uint8_t’ has not been declared
   29 |         void set_rgb(uint8_t n, uint8_t vr, uint8_t vg, uint8_t vb)
      |                      ^~~~~~~
src/psteelseries.h:29:33: error: ‘uint8_t’ has not been declared
   29 |         void set_rgb(uint8_t n, uint8_t vr, uint8_t vg, uint8_t vb)
      |                                 ^~~~~~~
src/psteelseries.h:29:45: error: ‘uint8_t’ has not been declared
   29 |         void set_rgb(uint8_t n, uint8_t vr, uint8_t vg, uint8_t vb)
      |                                             ^~~~~~~
src/psteelseries.h:29:57: error: ‘uint8_t’ has not been declared
   29 |         void set_rgb(uint8_t n, uint8_t vr, uint8_t vg, uint8_t vb)
      |                                                         ^~~~~~~
src/psteelseries.h:36:23: error: ‘uint8_t’ has not been declared
   36 |         void set_mode(uint8_t n, uint8_t val)
      |                       ^~~~~~~
src/psteelseries.h:36:34: error: ‘uint8_t’ has not been declared
   36 |         void set_mode(uint8_t n, uint8_t val)
      |                                  ^~~~~~~
src/psteelseries.h:44:9: error: ‘uint8_t’ does not name a type
   44 |         uint8_t r       = 0x00; // ??
      |         ^~~~~~~
src/psteelseries.h:1:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
  +++ |+#include <cstdint>
    1 | /*
src/psteelseries.h:45:9: error: ‘uint8_t’ does not name a type
   45 |         uint8_t g       = 0x00; // ??
      |         ^~~~~~~
src/psteelseries.h:45:9: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
src/psteelseries.h:46:9: error: ‘uint8_t’ does not name a type
   46 |         uint8_t b       = 0x00; // ??
      |         ^~~~~~~
src/psteelseries.h:46:9: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
src/psteelseries.h:47:9: error: ‘uint8_t’ does not name a type
   47 |         uint8_t f03     = 0x00; // ??
[... 696 lines cut ...]
   59 | struct steel_data_0e
      |        ^~~~~~~~~~~~~
src/psteelseries.h:59:8: note:   candidate expects 0 arguments, 3 provided
src/psteelseries.h:59:8: note: candidate: ‘constexpr steel_data_0e::steel_data_0e(const steel_data_0e&)’
src/psteelseries.h:59:8: note:   candidate expects 1 argument, 3 provided
src/psteelseries.h:59:8: note: candidate: ‘constexpr steel_data_0e::steel_data_0e(steel_data_0e&&)’
src/psteelseries.h:59:8: note:   candidate expects 1 argument, 3 provided
src/psteelseries.h:519:58: error: no matching function for call to ‘steel_data_0e::steel_data_0e(int, int, int&)’
  519 |                 write_led_rec(steel_data_0e(0x28, 0x3e, v));
      |                                                          ^
src/psteelseries.h:59:8: note: candidate: ‘steel_data_0e::steel_data_0e()’
   59 | struct steel_data_0e
      |        ^~~~~~~~~~~~~
src/psteelseries.h:59:8: note:   candidate expects 0 arguments, 3 provided
src/psteelseries.h:59:8: note: candidate: ‘constexpr steel_data_0e::steel_data_0e(const steel_data_0e&)’
src/psteelseries.h:59:8: note:   candidate expects 1 argument, 3 provided
src/psteelseries.h:59:8: note: candidate: ‘constexpr steel_data_0e::steel_data_0e(steel_data_0e&&)’
src/psteelseries.h:59:8: note:   candidate expects 1 argument, 3 provided
src/msigd.cpp: In function ‘int steel_main(std_logger_t&, int, char**)’:
src/msigd.cpp:1371:41: error: invalid types ‘int[int]’ for array subscript
 1371 |                 profile_data[i].col[0].e[0] = i;
      |                                         ^
make: *** [makefile:120: src/msigd.o] Error 1

```

NostalgiaRealm, 2025-01-10:
For anyone coming across this: I just found out that in Fedora Linux 41 KDE (probably also on the GNOME spin) you can just change the monitor's brightness from the brightness slider in the panel. If that's all you need, then you might no longer need msigd.
[image]

dmsft, 2025-05-07:
@NostalgiaRealm 
Installing `libusb-compat-0.1` and `libusb-compat-0.1-devel` allowed me to compile without HIDAPI.

```shell
sudo dnf install libusb-compat-0.1 libusb-compat-0.1-devel
make USE_HIDAPI=0
```

NostalgiaRealm, 2025-05-07:
> [@NostalgiaRealm](https://github.com/NostalgiaRealm) Installing `libusb-compat-0.1` and `libusb-compat-0.1-devel` allowed me to compile without HIDAPI.
> 
> sudo dnf install libusb-compat-0.1 libusb-compat-0.1-devel
> make USE_HIDAPI=0

That's great! Hopefully the README can be changed to reflect the changes in the package's names.